MarlinFirmware/Marlin #18375: [BUG] U8glib-HAL branch merged

richbria, 2020-06-21:
### Bug Description

As of 5 hours ago the bugfix branch on U8glib-HAL was merged in per https://github.com/MarlinFirmware/U8glib-HAL/pull/18 .   

Since this is referenced in https://github.com/MarlinFirmware/Marlin/blob/2.0.x/platformio.ini using the branch name, this breaks the Marlin build.

### My Configurations

I am using the default config for an Ender 3 Pro

### Steps to Reproduce

Try to build using Marlin Firmware Auto Build.

**Expected behavior:** 
No Errors Fetching Dependencies

**Actual behavior:** [What actually happens]
`FDUnrecognizedStatusCode: Got an unrecognized status code '404' when downloaded https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip`

#### Additional Information
Initial issue report:
https://github.com/MarlinFirmware/U8glib-HAL/issues/19

I forked and updated the links and was able to get past the dependency issues using a Sanguino1284p (Ender 3 Pro).  I am by no means an expert on any of this.  Today is my first time ever interacting with this Firmware so take this with a grain of salt.

ellensp, 2020-06-21:
URL https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip works, no 404 error.

thisiskeithb, 2020-06-21:
> URL https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip works, no 404 error.

The branch was restored, so it works now 🙂

boelle, 2020-06-21:
can we safely close this one then?

thisiskeithb, 2020-06-21:
> can we safely close this one then?

Yes

closed 2020-06-21 by boelle