MarlinFirmware/Marlin #18381: [BUG] Can't home in any axis :( (Sidewinder x1)

nicedevil007, 2020-06-21:
Heya,

I'm using the latest commits from today with marlin bugfix release.
I used the config that is provided at the other Marlin repository. The MKS board is exchanged with a SKR 1.4 turbo board.

TMC2209's are used. Sensorless homing isn't activated right now.

While the printer is homing each axis the printer only mvoes the axe 2mm in one direction and then stops. Doesn't matter what axe is homed. The printer will do it on every axis.

M119 shows that all endstops are trigger :( That will happen when the printer is sitting anywhere.

thisiskeithb, 2020-06-21:
Since you’re not using sensorles homing, did you cut the diag pins on the drivers? They won’t work with endstops otherwise.

nicedevil007, 2020-06-21:
that makes sense... forget it :D

sjasonsmith, 2020-06-21:
I find that to be a terrible design choice on these boards. Requiring a user to permanently modify driver modules because they don't want to enable sensorless homing is very annoying. There are useful features that could be implemented using the sense pin even if not used for homing, but permanently attaching them to endstop pins makes that harder to achieve.

Anyway, I'll close this out since it does not seem to be a Marlin bug.

closed 2020-06-21 by sjasonsmith

thisiskeithb, 2020-06-21:
> I find that to be a terrible design choice on these boards. Requiring a user to permanently modify driver modules because they don't want to enable sensorless homing is very annoying.

I wish they’d bring back jumpers for controlling sensorless homing like on the SKE 1.3, but it is what it is 🙂