MarlinFirmware/Marlin #18387: @astro-smart still an issue? Extruder motor stops spinning during any print

astro-smart, 2020-06-21:
@astro-smart still an issue?

_Originally posted by @boelle in https://github.com/MarlinFirmware/Marlin/issues/18261#issuecomment-647062190_

I did notice a new issue with this firmware and with an MKS-GEN-L board version 1 with a Ender -3.  For some reason, when you start a print, the extruder motor stops spinning after about 30 secs or so in the print. Manual extrusion using the menu and G1 commands work fine but not for a print with any file. If I change the firmware back to 1.1.9, everything works fine. Any suggestions?

ellensp, 2020-06-22:
Please follow issue template. 
When starting a new issue you need to attach Current Configuration files.

sjasonsmith, 2020-06-22:
Often the extruder motor stopping is related to linear advance with TMC stepper drivers. There are several open issues related to this.

I'm closing this issue, since it seems like a duplicate of many existing issues.
https://github.com/MarlinFirmware/Marlin/issues?q=is%3Aissue+is%3Aopen+%22extruder+stops%22

Please look through those other issues, to see if they can help answer any of your questions.

closed 2020-06-22 by sjasonsmith