MarlinFirmware/Marlin #18390: [BUG] Vertical lines on printed part

MarkMan0, 2020-06-23:
### Bug Description

There are vertical lines on the printed part. It's best described as the beginning of corners(ringing), but in my case, it's the same along the whole x-y axes - in case of a cube. If I print a cylinder, the lines are visible everywhere.
Picture in comments

The "amplitude" of the lines changes based on speed.

I've tried to print the same gcode with Marlin 1.1.9 and a Rumba board, and the lines weren't present, so I believe it's a software issue.

### My Configurations

Custom printer:
+ Cartesian printer, with ball screws
+ Board: SKR PRO
+ Marlin: 2.0.x(also present in 2.0.0)
+ Drivers: TMC2130 on XYE, TMC5160 on Z
+ Prints started from SD card, with large buffer size

**Expected behavior:** 
Print without vertical lines

**Actual behavior:**
See picture

#### Additional Information
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4821318/Marlin.zip)

boelle, 2020-06-23:
what version are you using?

[bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) is the one to use

MarkMan0, 2020-06-23:
2.0.x, going to try bugfix now

closed 2020-06-24 by thinkyhead