MarlinFirmware/Marlin #18391: [BUG] G28 Y does not work with CoreXY and sensorless homing

Minims, 2020-06-23:
### Bug Description

G28 Y does not work with CoreXY and sensorless homing
G28 X is OK
G28 Z is OK

### My Configurations

SKR Pro 1.1
Core XY
TMC2209
XY SensorLess Homing 

[CoreXY-SKR-Pro.zip](https://github.com/MarlinFirmware/Marlin/files/4821453/CoreXY-SKR-Pro.zip)
### Steps to Reproduce

1. G28 X, confirm it's OK
2. G28 Y, No move on Y

**Expected behavior:** [What you expect to happen]

G28 Y works, (Y goes to Y Min) 

**Actual behavior:** [What actually happens]

G28 Y not working, (Y does not move) 

#### Additional Information

Works on a SKR 1.4 Turbo, 2209, SensorLess Homing, but it's a Prusa, not a CoreXY.

boelle, 2020-06-23:
what version of marlin?

Minims, 2020-06-23:
Latest bugfix-2.0.x (compile 2 hours ago)

thinkyhead, 2020-06-23:
Duplicate of #18212

closed 2020-06-23 by thinkyhead