MarlinFirmware/Marlin #18397: [BUG] Pause command request a click button after G4

ils15, 2020-06-24:
I tried several times to use a pause command in my end gcode
But printer request a action

[image]

Tried several things like disable emergency phaser, pins, etc

Motherboard - MKS Robin E3D
Latest Marlin Bugfix

ellensp, 2020-06-24:
Please provide Configuration files.

thisiskeithb, 2020-06-24:
This issue was already closed under your other username: https://github.com/MarlinFirmware/Marlin/issues/18393

thisiskeithb, 2020-06-24:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

* MarlinFW.org [Marlin Documentation](http://marlinfw.org)
* RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
* [Tom's 3D Forums](https://discuss.toms3d.org/)
* Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
* Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
* [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
* Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-06-24 by thisiskeithb