MarlinFirmware/Marlin #18398: [BUG] Extruder temperature 10ºC below setpoint

pablopeu, 2020-06-24:
I'm using latest bugfix on my anycubic delta kossel plus (config: [Marlin-BugFix_June_9.zip](https://github.com/MarlinFirmware/Marlin/files/4822791/Marlin-BugFix_June_9.zip)) did the autotune per marlin recomendations, rewrote the PID values and reuploaded the firmware.

When printing, first layer without fan (M107) temperature dithers very close to the set temperature, but when the second layer starts it goes down and oscilates around 10ºC less than the set point. This can be clearly be seen in octoprint graph:

[image]

Second layer with the fan on (M106) temperature increases slightly but after the third layer it remains at around 10ºC less until the end of the print with the fan always on.

What is the problem? should I calibrate PID with the fan on? or its something else?

Just in case, this is the gcode: [OttoDIYH_head_v07.zip](https://github.com/MarlinFirmware/Marlin/files/4822846/OttoDIYH_head_v07.zip)

pablopeu, 2020-06-24:
After two hours into the print, and veeeeeeeery slowly now its only 5C below setpoint.

ellensp, 2020-06-24:
This looks like your fans are blowing on the hotend itself.
There should be two fans, one that blows on the cooling fins of the hotend. The second which blows on the freshly extruded plastic. 
Neither should blow on the hotend.

pablopeu, 2020-06-24:
There are two, you are right, a turbine which blows down and with the original deflector heats the layer just printed and the one that blows to the fin. In a nutshell its the original unchanged hotend. 
Can try to print a new deflector and check if it improves.

closed 2020-06-24 by thinkyhead