MarlinFirmware/Marlin #18404: [BUG] (short description)

olovskos, 2020-06-24:
### Bug Description

When using a radds board (with a due of course) and setting the homing direction to 1 it breaks the firmware and it behaves weird.

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/4826868/Config.zip)

### Steps to Reproduce

1. Download Marlin2 release
2. Set Z_HOME_DIR 1 and #define USE_ZMAX_PLUG
3. Compile and Upload

**Expected behavior:** [What you expect to happen]
The board is working as supposed

**Actual behavior:** [What actually happens]
When moving the z axis with G0 Z1 for example, the z axis moves in one direction until the power is disconnected from the system. It even doesn`t matter if you set G0 Z to a positive or negative value, it always moves in the same direction for ever. 

#### Additional Information

If the settings are default (Z_HOME_DIR -1 and //#define USE_ZMAX_PLUG) everything is working but that is not the setting I need.

shitcreek, 2020-06-26:
The configs you upload needs to be the one in question otherwise we wouldn't know what isn't working.
The config is using `BOARD_BTT_SKR_V1_4` and not `BOARD_RADDS`
When you enabled `USE_ZMAX_PLUG`, did you disable `USE_ZMIN_PLUG`? Did you you plug the endstop into the ZMAX pin?
Are you certain this is a bug and not configuration error? Please exhaust these other sources first:
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report. Thanks.

olovskos, 2020-06-28:
> The configs you upload needs to be the one in question otherwise we wouldn't know what isn't working.
> The config is using `BOARD_BTT_SKR_V1_4` and not `BOARD_RADDS`

Where did you find "BOARD_BTT_SKR_V1_4" ? In the Configuration.h Line 131 it is "#define MOTHERBOARD BOARD_RADDS". 

> When you enabled `USE_ZMAX_PLUG`, did you disable `USE_ZMIN_PLUG`? Did you you plug the endstop into the ZMAX pin?

Yes and Yes.

> Are you certain this is a bug and not configuration error?

It clearly seems like a bug to me. If I set "Z_HOME_DIR 1" and try to compile, an error is thrown that I have to enable "USE_ZMAX_PLUG". Sounds right, so I enabled "USE_ZMAX_PLUG"and disabled "USE_ZMIN_PLUG". Now the firmware compiles without errors. If there should be any further settings that need to be changed, the compiler messages/errors would tell me or not? Why would the firmware behave this strange when only the homing direction is changed? Are there any other settings that need to be changed when someone tries to change the homing direction?

olovskos, 2020-07-30:
So?

boelle, 2020-07-31:
> The configs you upload needs to be the one in question otherwise we wouldn't know what isn't working.

boelle, 2020-07-31:
> The config is using `BOARD_BTT_SKR_V1_4` and not `BOARD_RADDS`

boelle, 2020-07-31:
> When using a radds board (with a due of course)

boelle, 2020-07-31:
have you checked that endstops are working ? M119 with a finger on them and then M119 again without a finger on them

at power up marlin think that nozzle is at 0,0,0

what happens when you send G0 X10, does it move away from endstop?
same for Y ?

olovskos, 2020-07-31:
> The configs you upload needs to be the one in question otherwise we wouldn't know what isn't working.

Still don`t know what that means? Do I have to set the config to "BOARD_BTT_SKR_V1_4" when using a radds board or what do you mean?

In this/my question right here https://github.com/MarlinFirmware/Marlin/issues/18404 when you download the Config.zip, you open it, you open the Configuration.h, line 131 it is defined "#define MOTHERBOARD BOARD_RADDS"... At least when I download it...

> what happens when you send G0 X10, does it move away from endstop?

When sending ANY movement command, regardless of X,Y or Z, only the Z Axis moves in one direction until the power is disconnected.

> have you checked that endstops are working ? M119 with a finger on them and then M119 again without a finger on them at power up marlin think that nozzle is at 0,0,0

Yes, same behavior for both cases

boelle, 2020-07-31:
> Do I have to set the config to "BOARD_BTT_SKR_V1_4" when using a radds board

No set it to Radds

boelle, 2020-07-31:
> Still don`t know what that means?

config is not set to radds

boelle, 2020-07-31:
> When sending ANY movement command, regardless of X,Y or Z, only the Z Axis moves in one direction until the power is disconnected.

check endstop status with M119

closed 2020-09-05 by github-actions[bot]