MarlinFirmware/Marlin #18405: [BUG] Compile crashing on enable CUSTOM_STATUS_SCREEN_IMAGE

drackp2m, 2020-06-24:
### Bug Description
When try to activate option CUSTOM_STATUS_SCREEN_IMAGE the compilation of the error indicated at the end of the issue

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4826863/Marlin.zip)

### Steps to Reproduce
1. Uncomment option CUSTOM_STATUS_SCREEN_IMAGE
2. Add _Statusscreen.h to your Marlin folder
3. Try to compile

**Expected behavior:** Compile successfully

**Actual behavior:** The error inserted bellow

#### Additional Information
```
Compiling .pio/build/LPC1769/src/src/lcd/dogm/u8g_dev_tft_320x240_upscale_from_128x64.cpp.o
Marlin/src/lcd/dogm/status_screen_DOGM.cpp: In static member function 'static void MarlinUI::draw_status_screen()':
Marlin/src/lcd/dogm/status_screen_DOGM.cpp:519:78: error: expected ')' before 'u8g'
  519 |     if (PAGE_CONTAINS(STATUS_LOGO_Y, STATUS_LOGO_Y + STATUS_LOGO_HEIGHT - 1))
      |                                                                              ^
      |                                                                              )
  520 |       u8g.drawBitmapP(STATUS_LOGO_X, STATUS_LOGO_Y, STATUS_LOGO_BYTEWIDTH, STATUS_LOGO_HEIGHT, status_logo_bmp);
      |       ~~~                                                                     
In file included from Marlin/src/lcd/dogm/dogm_Statusscreen.h:32,
                 from Marlin/src/lcd/dogm/status_screen_DOGM.cpp:32:
Marlin/src/lcd/dogm/ultralcd_DOGM.h:219:31: note: to match this '('
  219 | #define PAGE_CONTAINS(ya, yb) ((yb) >= u8g.getU8g()->current_page.y0 && (ya) <= u8g.getU8g()->current_page.y1) // Do two vertical regions overlap?
      |                               ^
Marlin/src/lcd/dogm/status_screen_DOGM.cpp:519:9: note: in expansion of macro 'PAGE_CONTAINS'
  519 |     if (PAGE_CONTAINS(STATUS_LOGO_Y, STATUS_LOGO_Y + STATUS_LOGO_HEIGHT - 1))
      |         ^~~~~~~~~~~~~
Marlin/src/lcd/dogm/status_screen_DOGM.cpp:520:112: error: expected ')' before ';' token
  520 |       u8g.drawBitmapP(STATUS_LOGO_X, STATUS_LOGO_Y, STATUS_LOGO_BYTEWIDTH, STATUS_LOGO_HEIGHT, status_logo_bmp);
      |                                                                                                                ^
      |                                                                                                                )
Marlin/src/lcd/dogm/status_screen_DOGM.cpp:519:8: note: to match this '('
  519 |     if (PAGE_CONTAINS(STATUS_LOGO_Y, STATUS_LOGO_Y + STATUS_LOGO_HEIGHT - 1))
      |        ^
*** [.pio/build/LPC1769/src/src/lcd/dogm/status_screen_DOGM.cpp.o] Error 1
```

#### Extra additional Information
When using the [Bitmap to C/C++ Converter](https://marlinfw.org/tools/u8glib/converter.html) tool, there is an option to align the status screen to the right, however this option never worked for me.

Should I open another issue to see if anyone knows how to fix that problem?

**BTW**, because my current computer is a bit grungy and slow, I can't test every firmware version, but I'd say it's been a thing of the last week (or short time ago).
Before the last changes, the status screen compiled correctly, however, as I mentioned before, I never managed to get it aligned right.

As always, thanks for everything.

ellensp, 2020-06-25:
Confirmed on latest bugfix.

thisiskeithb, 2020-06-25:
Adding `#define STATUS_SCREEN_Y 0` to `_Statusscreen.h` allows it to compile, but I haven't tested it on real hardware.

ellensp, 2020-06-25:
Yes bug is in autogenerator vs marlin itself. 
current eg all have a headers of the form
#define STATUS_LOGO_X            8
#define STATUS_LOGO_Y            5
#define STATUS_LOGO_WIDTH       22
vs the provided header of
#define STATUS_SCREEN_X 96
#define STATUS_LOGO_WIDTH 32

drackp2m, 2020-06-25:
Perfect, I have added the definitions that you have told me and everything looks much better, and without giving compilation errors.

Now, my file this contains the following:
```
#define STATUS_LOGO_X         96
#define STATUS_LOGO_Y          0
#define STATUS_LOGO_WIDTH     32
#define STATUS_HEATERS_X       6
#define STATUS_BED_X          32
#define STATUS_FAN_X          64
```

And it looks like this:
[image]

Thank you all for your solutions, and I hope that the bug will be closed soon, which in principle is only to add the X and Y positions of the logo by default to zero.

ellensp, 2020-06-30:
looking at this again,  it is a bug in marlin, triggered when STATUS_LOGO_Y is not defined
```
   #ifndef STATUS_LOGO_Y 
     #define STATUS_LOGO_Y _MIN(0U, ((20 - (STATUS_LOGO_HEIGHT)) / 2) 
   #endif 
```
defines STATUS_LOGO_Y to a function _MIN(0U, ((20 - (STATUS_LOGO_HEIGHT)) / 2)
This has unmatched brackets , 4 opening brackets, but only 3 closing.

i believe its meant to be _MIN(0U, ((20 - STATUS_LOGO_HEIGHT)) / 2)

Going to do some testing.

closed 2020-07-01 by ellensp