MarlinFirmware/Marlin #18406: [FR] Verbose Z Steppers Auto-Alignment

drackp2m, 2020-06-24:
### Description
Similar to when you do a bed leveling, I think it would be interesting if it told you the number of the attempt it is executing, and when it has finished and goes to the next, indicate the last deviation obtained.
Because I have been monitoring these processes many times, and my final conclusion is that one cannot very much trust that in five attempts the motors are perfectly aligned.

### Feature Workflow
1. Execute G34 gCode
2. Display **Iteration 1/n...**
3. Wait for first sync try
4. Update display info with **Iteration 2/n... Diff 0.06**
5. Wait for second sync try
6. Repeat steps 4-5

sjasonsmith, 2020-06-25:
@drackp2m, please enable `DEBUG_LEVELING_FEATURE` feature in your `Configuration.h` file, and follow the instructions in the nearby comments to enable it at runtime.

With this enabled, the output from `G34` is _extremely_ noisy, but I think the details you are asking for will be buried in that output.

I would appreciate seeing examples of the current output (both with and without DEBUG_LEVELING_FEATURE), along with examples of what you would _like_ to see.

InsanityAutomation, 2020-07-22:
If im understanding this correctly, hes just looking for the display to get some of the strings already going out to serial.

drackp2m, 2020-07-22:
> If im understanding this correctly, hes just looking for the display to get some of the strings already going out to serial. If thats the case, ill keep this in mind as im working on some additions to this gcode anyway.

Exactly, or at least something similar.

The information that appears by series is perfect.

The information that appears on the screen is non-existent.

At a minimum, it would be nice if the final level of precision with which the various Z engines were leveled appeared.

sjasonsmith, 2020-07-22:
The screen did it even occur to me. The only printer I can use this on has no screen so I honestly had no idea It was even on a menu!

drackp2m, 2020-07-22:
> The screen did it even occur to me. The only printer I can use this on has no screen so I honestly had no idea It was even on a menu!

It is true, I did not mention the LCD screen either because it seemed obvious to me, since by console it did give a lot and very good information.

So I apologize for not clarifying the matter in advance. And I encourage you to add some information about it to the LCD, since I have the functionality enabled on my printer, I have a screen and I have a shortcut to the G34 command in the menus of my LCD. And I am very angry to have to go to the computer or mobile to monitor from Octoprint how aligned my two Z engines were after the first streak of attempts (which in the Firmware I have defined five)

InsanityAutomation, 2020-07-22:
Easy enough! Ive assigned this one to myself and I intend to be working in that file this weekend. Its almost nothing to do next to the other changes im working on.

drackp2m, 2020-07-22:
How good! What a joy!! I haven't touched the printer in days, but I really want to flash that update.

I hope you have a nice MR. Hehehehe.

Thanks for everything guys. <3

drackp2m, 2020-11-04:
I was there for many months without being able to touch my printer, and after all this time I really wanted to install the latest version of Marlin and test how these ads looked.
And it is good to know which iteration it is going through, but it would be interesting if at the end of each one of these it indicated the level of adjustment that has been obtained, for example:
`G34 Iteration: 2 (0.24)`

Or at least, that in the last one it will show a message like:
`Accuracy Achieved (0.02)`

However, they want to add the level of detail explained or decide to leave it as it currently is. I think the feature request could be closed now.

closed 2024-04-10 by InsanityAutomation