MarlinFirmware/Marlin #18412: [BUG] Z-Axis Motors seizing with 2.0 and later builds

T-SquaredProductions, 2020-06-25:
### Bug Description

I have tried so many configurations with the wiring, but I cannot get this to work properly with my Creality CR-10 S5. I don't know if it is the pin configuration or what.

### My Configurations

[Creality Config.zip](https://github.com/MarlinFirmware/Marlin/files/4828610/Creality.Config.zip)

### Steps to Reproduce

1. Upload to Creality CR-10 S5

**Expected behavior:** Motors smoothly move Z-Axis up and down from "Move Axis" menu

**Actual behavior:** 
(Depends, any of the 3 can happen:)
1. Motor grinds, then no further motion
2. Motor sounds good, but does not move
3. Motor makes noise of force (thunk), but no motion.

#### Additional Information

I have confirmed that the windings are good in my motors. The previous motors that I used had this problem as well, but their windings seemed weak.

Video Example: [Stuck Z-Axis Motors.mp4.zip](https://github.com/MarlinFirmware/Marlin/files/4828673/Stuck.Z-Axis.Motors.mp4.zip)

sjasonsmith, 2020-06-25:
@T-SquaredProductions have you tested this using the `bugfix-2.0.x` branch? This is where all fixes are made, so the problem must be reproduced from there prior to any further investigation.

Are you using the silent motherboard sold by Creality, or is this a stock (loud) motherboard?

T-SquaredProductions, 2020-06-25:
Thanks. I'll see what I can do.

> Are you using the silent motherboard sold by Creality, or is this a stock (loud) motherboard?

Well, I had problems with multiple boards. A RAMPS 1.4 board gave me similar problems with the X-Axis (I believe).

I'm using a board labeled as "Creality 2.2" (probably a silent one, because I don't know the difference), because I had similar problems with the stock 2.1 board. (Plus, I think I fried one of its driver chips when I unplugged the motor when energized while testing it.)

T-SquaredProductions, 2020-06-26:
Nope. Even using the bug-fix version didn't solve it.
[Marlin Bugfix 2.0.x Config.zip](https://github.com/MarlinFirmware/Marlin/files/4834927/Marlin.Bugfix.2.0.x.Config.zip)

shitcreek, 2020-06-26:
Have you tried adjusting the vref on the board?

thinkyhead, 2020-06-26:
Try reducing your Z max feedrate and your Z acceleration to very, very conservative values. Don't forget to use M502 (if changing them in your configs). See if you can find any range of max feedrate / acceleration where they move reliably.

T-SquaredProductions, 2020-06-27:
> Have you tried adjusting the vref on the board?

> Try reducing your Z max feedrate and your Z acceleration to very, very conservative values. Don't forget to use M502 (if changing them in your configs). See if you can find any range of max feedrate / acceleration where they move reliably.

I tried both. It still vibrates.

T-SquaredProductions, 2020-06-27:
I'm not sure why this is happening, but the microstepping doesn't seem to work, or it's skipping. I don't get a smooth turn with the "Move 0.1 mm" setting. It seems to ignore some steps, and move backwards with others.

T-SquaredProductions, 2020-06-28:
Changing the feedrate (down to 1) just produces the same kind of glitch (not turning properly), except the turning of the motor is much less jittery.

Acceleration does the same thing, except the jittering starts slow, then up to fast motion, and then slow again.

T-SquaredProductions, 2020-06-28:
Ok, I decided to ditch the Creality board in favor of a Brand-New BigTreeTech SKR 1.4. Still the same problems.
At this point I'm believing it more programming error than bad wiring. I could try the acceleration and feedrate again, but I'm not so sure they'll help. (Yes, I know that's a defeatist attitude, but I have tried using Marlin 2.0 on FOUR boards without success.)

T-SquaredProductions, 2020-06-29:
Ok, after looking up some info, I had my jumpers wrong on all the drivers. I now have smooth movement back on X & Y, but Z still moves erratically, and in addition, have some strange noise (as in, it sounds like TV static) from the Z motor that I am using to test.

T-SquaredProductions, 2020-07-01:
I was wrong. I _was_ overdriving my Z driver, because the heatsink was burning hot. Replacing the driver and updating the firmware fixed everything.

closed 2020-07-01 by T-SquaredProductions