MarlinFirmware/Marlin #18413: [BUG] MKS robin Lite v1.1 freezing at 100% and won't update again

theprintsurgeon, 2020-06-25:
### Bug Description

This issue was addressed previously with someone else but now it is happening again. With the latest marlin bug fix downloaded 6/24/20 the board will not update and freezes on 100%. I have the MKS Robin Lite board v1.1 with the MKS robin lite LCD 12864 v1.1 screen. I have tried all combinations of serial ports and screens. If someone can just provide me with a working project folder for this board screen combo that would be great.

### My Configurations

[config files.zip](https://github.com/MarlinFirmware/Marlin/files/4828959/config.files.zip)

### Steps to Reproduce

1. Compile and load on SD
2. Insert SD
3. Update

**Expected behavior:** Complete update, restart, function normally

**Actual behavior:** Freezes at 100%

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-06-30:
I would not even know where to begin with this one, if it is a problem with the actual firmware update process?

@thisiskeithb, are you familiar with this board, and anything that might cause this behavior?

thisiskeithb, 2020-06-30:
> @thisiskeithb, are you familiar with this board, and anything that might cause this behavior?

I am not.

theprintsurgeon, 2020-06-30:
Hi ! Thanks for checking in on this. It's definitely something with the stm32 boards. I basically just swapped out the board for another one. But I wasn't able to get any of the MKS Robin firmwares to compile because of various errors. In most of these cases I left the settings default and just changed the board name or the board and the screen.

sjasonsmith, 2020-06-30:
The original issue said it couldn’t update to the board after you built the firmware. Your last post indicates you could not compile successfully. Which is it?

If it is the original issue with the firmware update, what do you mean by “it gets to 100%”?  Where do you see that?

theprintsurgeon, 2020-06-30:
Right, sorry, let's just focus on one issue at a time. there was a severe problem trying to get the firmware to compile in the first place. Just selecting that bored I mention that because it's the easiest to start with and reproduce. 

Now, when I actually do get the firmware to compile and I upload it to the board the MKS screen on the device will show that it's updating, get to 100%, and stay there. I will try to power on the board and take a picture shortly.

theprintsurgeon, 2020-06-30:
Here is the update issue I am referring to. It just hangs on this screen. This is the most recent nightly build of Marlin. 

As a separate issue that I am just mentioning so you know what troubleshooting steps I've already taken: the stable marlin release causes a boot loop, it updates displays the Marlin bigfix screen, then the custom splash screen that I have added then restarts indefinitely.

I have also gone to the makerbase page and downloaded the marlin firmware that they provide. I am unable to compile this firmware. I have tried on two computers. They provide a factory firmware.Bin file. If I use this file the board will update but I can't use it because I have no idea what features are enabled or what printer volume, speeds, etc they built into their published .bin file. 
[image]

sjasonsmith, 2020-07-31:
The challenge here is that your board seems to be uncommon and maybe now impossible to buy. This makes it hard for anybody to reproduce the issue. I was hoping someone else with an MKS Lite v1.1 would notice this  and be able to help.

theprintsurgeon, 2020-07-31:
> The challenge here is that your board seems to be uncommon and maybe now impossible to buy. This makes it hard for anybody to reproduce the issue. I was hoping someone else with an MKS Lite v1.1 would notice this and be able to help.

I can mail you the board if that helps. It's useless to me its current state

closed 2020-09-06 by github-actions[bot]