MarlinFirmware/Marlin #18415: [BUG] Temperature reach no more working after update

spilz87, 2020-06-25:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

Hello,

I was running on 852a8d6685ec1137eb65e78fa748cae41fbd36b6 
I print at 260°C everything worked well.

I updates to c3d2ffc09c8a15398894bb2050397b9653878923
EEPROM was reset as they are no more compatible.
I keep the same PID settings for hotend 

no more possible to stay at 260°C : it heat until 267°C then cool down slowly to 250°C and is no more able to go back to 260°C

I tryed to get new PID values with M303 (PID autotune) but than doesnt solved the issue :(

**Expected behavior:**

be able to stay at 206°C to print as before

**Actual behavior:**

Not able to stay at 260°C, so error occures and printer stop working

Is there a modification in PID and temperature asservicement between both version ?
is there anyone getting the same issue ?

Thanks for your help

boelle, 2020-06-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

spilz87, 2020-06-30:
[Configuration and _adv.zip](https://github.com/MarlinFirmware/Marlin/files/4851794/Configuration.and._adv.zip)
thanks

spilz87, 2020-06-30:
Im not able to stay at 260°c, it Always go from 250 to 270 again and again

shitcreek, 2020-07-02:
Have you given bugfix-2.x a try? There has been a few changes/fixes to the temperature codes in the past day: https://github.com/MarlinFirmware/Marlin/pull/18465

thinkyhead, 2020-07-03:
When you send `M301` from the host does it report back the PID values that you expect?

I heard from someone recently that perhaps the EEPROM read / write of PID values has been wonky.

I've been tweaking PID code in the last couple of days, so please give the latest `bugfix-2.0.x` a try.

spilz87, 2020-07-04:
I completely change my PID and going back to default, now it works ...

I find a solution but I don’t think it’s normal that upgrade change this ...

Can be closed if I’m the only one notice this 

Thanks for your help

spilz87, 2020-07-04:
When using M301 I got a close PID to the original one , not sure M301 really “explore all possibilities” for values, seems it stay close to the saved values

AnHardt, 2020-07-06:
M301 does not make use of any previous values. It always starts from zero, as if it had never run before. Because of the always varying environment (temperature, air speeds) it finds always similar but not identical values even when using always the same algorithm with the same start values.

closed 2020-07-12 by thinkyhead