MarlinFirmware/Marlin #18418: [BUG] Y Endstop event not triggered with Dual Y endstops and Home to Y max

c3D-Dan, 2020-06-25:


ellensp, 2020-06-25:
Please provide full actually used Configuration files. So we can drop in appropriate hardware and verify issue

c3D-Dan, 2020-06-25:
Here you go. 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4831095/Marlin.zip)

robbycandra, 2020-07-01:
It is because : 
- #define Y_HOME_DIR 1
- #define Y2_USE_ENDSTOP YMAX

boelle, 2020-07-02:
@ellensp @robbycandra so this is a config issue and not a bug?

c3D-Dan, 2020-07-02:
I don’t agree.   I have tried inverting / disabling those options with no success at all.  Best I can get is one endstop to trigger which makes the other y motor smash into the other endstop.  My Y end stops have to be at Y max for logistical reasons.

c3D-Dan, 2020-08-02:
Problem still remains

c3D-Dan, 2020-09-08:
This should not be closed.  It's a simple issue, the only reason more don't comment on it is because it only occurs with rear left hand co ordinate systems.  Please reopen.

sjasonsmith, 2020-09-08:
Unfortunately it’s not that easy for any dev to test, because most machines don’t have multiple Y motors. Someone would have to contrive a setup without having actual hardware to test with.

c3D-Dan, 2020-09-08:
I’d be happy to assist where possible.  A dev need just ask.

sjasonsmith, 2020-09-08:
So keep in mind that most devs are just random people like me that fix bugs in their spare time. Attention tends to go toward hardware similar to their own. If you have any ability to fix the problem and contribute a patch, that would be most welcome.

c3D-Dan, 2020-09-23:
Thanks for re opening!

sjasonsmith, 2020-09-23:
There is apparently some disagreement about what "stale" means.
I consider it to mean people are unresponsive and no longer care about finding a solution, which was certainly not your situation.

The bot worked exactly as intended in this case, unflagging the Bug when you replied. Unfortunately a human threw a wrench in the works.

c3D-Dan, 2020-09-23:
I understand. Many users don't provide updates when they figure their problem.  Likewise, my work around for this was not to use YMax, rather keep Y min combined with x max and just accept a rear left hand coordinate system it creates.  If we can get Y max working, I'll be a happy boy.

I think the moral here is to keep hounding for updates to prevent real issues from being marked stale. 

Dan

thisiskeithb, 2020-09-23:
> I think the moral here is to keep hounding for updates to prevent real issues from being marked stale.

Just comment when stale bot kicks in.

Someone will need the time, hardware, and interest in an issue they can test/fix and will reply when they find something.