MarlinFirmware/Marlin #18420: [FR] ADC tuning regarding the vcc input

ShadowOfTheDamn, 2020-06-25:
### Description

ADC needs to be tuned with regards to the input voltage
this feature should be implemented on the core of the marlin, regarding the analogue data reads like temperature sensor reads and so on. As you know Arduino boards does not have a reference voltage pin connected and so the controller will use the Vcc as the V_ref for ADC conversions. this will result in imprecise readings of the analogue input.
there is a video on the you-tube channel  "Andreas Spiess" with the following link: https://www.youtube.com/watch?v=UAJMLTzrM9Q
in this link it is explained why the ADC is imprecise on many controllers regarding the reference voltage sampling and a code is suggested for somehow correcting the current ADC result for super precise measurements
please consider this for more accurate temperature readings on all hardware.

#### Additional Information

# Video showing the way ADC work on ESP8266, ESP32 and Arduino micro-controllers as well as external ADC boards
https://www.youtube.com/watch?v=UAJMLTzrM9Q

# Video of the code in use for precise ADC measurements
https://www.youtube.com/watch?v=xI_qU2auVx8

# Sample code of the previous Video
https://github.com/SensorsIot/ADC_Test/

AnHardt, 2020-06-25:
That's right as long we are measuring resistor relations.
But there are a few cases where we are indeed measuring voltages directly. For example when we use tables 20 or 21,  -1 or -4. And the we **are** in trouble. There switching the power supply for the processor from USB to an other voltage regulator usually makes a noticeable difference.

AnHardt, 2020-06-30:
Additionally to the already provided solution for the AVRs i could find a [solution](https://github.com/stm32duino/wiki/wiki/API#analog) for the STM32s using stm32duino

AnHardt, 2020-06-30:
The best exactness could be achieved by handling the Vref-channel like the other ADC-pins, in every loop over the pins.  
A far less performance intensive variation could be to test Vref only one during startup. This would at least be able to handle the big differences when running from different sources for the power.