MarlinFirmware/Marlin #18426: [BUG] Build error in dwin.cpp

CRCinAU, 2020-06-26:
@thinkyhead - I think this is what you've been working on lately?

```
Marlin/src/lcd/dwin/dwin.cpp:2389:1: error: expected identifier before 'void'
 void HMI_Temperature(void) {
 ^~~~
Marlin/src/lcd/dwin/dwin.cpp:2389:26: error: two or more data types in declaration of 'HMI_Temperature'
 void HMI_Temperature(void) {
```

thisiskeithb, 2020-06-26:
Please follow the Issue Template when filing a bug report:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

CRCinAU, 2020-06-26:
Fixed in https://github.com/MarlinFirmware/Marlin/commit/452072cc08efcfdcd01c712c69cc81a014ea3626

Thanks @thinkyhead

closed 2020-06-26 by CRCinAU