MarlinFirmware/Marlin #18428: [FR] Add support for BTT SKR Mini E3 V2.0

swilkens, 2020-06-26:
Looks like SKR brought out a new version of the SKR mini E3 board for the Ender 3, now has additional fan control and a second Z-Axis connector, as well as an onboard AT24C32 eeprom.

**Manual:**
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0%20Instruction%20Manual.pdf

**PINS:**
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-PIN.pdf

**FIRMWARE:**
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0/Marlin-2.0.x-SKR-mini-E3-V2.0

thisiskeithb, 2020-06-26:
The Mini E3 V2 config is based on the prior SKR Mini configs (which are based off the stock Ender-3 config).

> It doesn't always print from when i select my print in the menu, i have to turn it off then back on and then it'll print

This is the first report I’ve heard of this happening. Have you modified any of the default Mini E3 config settings? If you could attach them for review, that would help.

> For some reason it doesn't raise the hotend when moving from multiple parts of a print, leading to blobs all over my prints.

This sounds like you need to enable Z hop/whatever the equivalent setting is called for your slicer since Marlin doesn’t control that.

sjasonsmith, 2020-09-11:
This board is supported and used by many people. I consider this feature request to be complete. If further issues are found, please refer to the resources below for help.

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-11 by sjasonsmith