MarlinFirmware/Marlin #18429: [FR] Dual X Carriage tool settings

DrumClock, 2020-06-26:
Would it be possible to add settings in configuratin_adv in the  `#define DUAL_X_CARRIAGE` section to determine which tool (extruder) is on X1 and X2?

e.g.
`  #define X1_CARRIAGE_TOOL {T0, T1}   // extruder E0 , E1 .....`
`  #define X2_CARRIAGE_TOOL {T2, T3}   // extruder E2 , E3 ..... `  

I would like to test a 4-material print.

InsanityAutomation, 2020-06-27:
This is on the list of things that have been discussed, with 3 tools on each for a max of 6 extruders in the works. We have hardware being worked up to develop with. So yes, but itll be awhile as hardware is still in development to work with. If a manufacturer is willing to contribute some to those of us working on it, it would speed things along greatly. Currently @Roxy-3D and myself are slowly chasing this with others looking at the board in development for other purposes as well.

DrumClock, 2020-06-27:
Hi, I'm not a manufacturer but I can almost contribute if it would move my request forward and I could start testing on my printer.
How should I contribute Roxy-3D and you?

InsanityAutomation, 2020-06-27:
Right now the primary constraint is the rest of the code for the board. Bootloader and HAL are in progress. Once that's done, we will need to go over mechanical design mounting a diamond hotend to a Trex 2+ carriage (the machine were both starting with) along with figuring out management for all 6 spools.

DrumClock, 2020-06-27:
Hi, which board will you use for testing?

InsanityAutomation, 2020-06-27:
[image]

Custom design Gary Marsh did for us. We have 5 prototypes atm with LPC 4078 chips

DrumClock, 2020-06-27:
I have a RUMBA board where I use all the HW.
I use dual X carriage stepper X1 and X2 and two servos
which I use for switching extruders E0 and E1 including nozzles.

All I need is to assign tools T0 + T1 for carrier X1 and tools T2 + T3 for carrier X2, and unfortunately this is not possible in the current Marlin. If I can, I can start testing the IDEX and HOTEND_OFFSET_X tool settings

Roxy-3D, 2020-06-27:
What you want to do isn't very much code.   But it is complicated enough that I don't want to attempt anything without having a machine to code against and test the changes.    The good news is, what you want to do is a subset of what we plan on doing.   What you want should fall out naturally.     The bad news is, it will probably be 3 or 4 months before we have something you can use.

InsanityAutomation, 2020-06-28:
I've got a GTR10 with expansion here and considered using it, but we all have alot of tasks going and efficiency in them is important! It'll get there, but unless there's financial incentive that portion isn't going to jump the queue.

thinkyhead, 2020-06-28:
Unless I am tripping, I think we have some sort of `SINGLENOZZLE` or Switching Nozzle support for Dual X Carriage at this time, which gets us most of the way, since I assume whatever is feeding each carriage is something like that.

InsanityAutomation, 2020-06-28:
Not currently unfortunately. Toolchange isn't smart enough to split extruders from physical tools yet. It's been discussed a few times so it's something were definitely moving towards.

DrumClock, 2020-06-28:
Hi,
The current Marlin can do the tool T0 to carrier X1 and T1 to X2.
Is it really a problem to insert the condition that T0 or T1 will be on X1 and the tool T2 or T3 will be on carrier X2?

Ziyayildizz3, 2020-07-01:
Hey! Ready for my machine to try this prompt. I was working on it and it's great to come across. I can try the software. His equipment is ready. I use 4 extruders and dualx systems. I tried to activate SINGLENOZZLE but I encountered an error. I want to try it soon. Yours.

DrumClock, 2020-07-02:
Thank you for your comments on the above feature. This closes the query.

DrumClock, 2020-07-03:
### This is my project 4 Extruders on IDEX

[image]

thinkyhead, 2020-07-03:
That looks like a `DUAL_X_CARRIAGE` plus `SWITCHING_EXTRUDER` plus `EXTRUDERS 4` setup. These options are intended to be usable together, but since that capability was first enabled we have not had a lot of testers and coders working on refining it.

DrumClock, 2020-07-03:
Hi, The subroutines were edited by Stephen. 
I'm not a programmer.  
I just have HW here and I'm testing

ElectroFlux-SS, 2020-07-05:
Hi this is Stephen.  The changes made can probably be made by just comparing the original Marlin files over the ZIP DrumClock provided, however listing below for the sake of time or that somebody can spot something quickly.

SanityCheck.h
   /**
 * A Dual Nozzle carriage with switching servo
 */
#if ENABLED(SWITCHING_NOZZLE)
  //#if ENABLED(DUAL_X_CARRIAGE)
  //  #error "SWITCHING_NOZZLE and DUAL_X_CARRIAGE are incompatible."
  //#elif ENABLED(SINGLENOZZLE)
#if ENABLED(SINGLENOZZLE)
    #error "SWITCHING_NOZZLE and SINGLENOZZLE are incompatible."
  #elif DISABLED(DUAL_X_CARRIAGE) && EXTRUDERS != 2
    #error "SWITCHING_NOZZLE requires exactly 2 EXTRUDERS."
  #elif DISABLED(DUAL_X_CARRIAGE) && NUM_SERVOS < 1
    #error "SWITCHING_NOZZLE requires NUM_SERVOS >= 1."
  #elif ENABLED(DUAL_X_CARRIAGE) && EXTRUDERS != 4
    #error "SWITCHING_NOZZLE along with DUAL_X_CARRIAGE requires exactly 4 EXTRUDERS."
  #elif ENABLED(DUAL_X_CARRIAGE) && NUM_SERVOS < 2
    #error "SWITCHING_NOZZLE along with DUAL_X_CARRIAGE requires NUM_SERVOS >= 2."
  #endif

tool_change.cpp and motion.cpp
   I would continue to use the original Marlin source files here.  No matter what changes I had attempted here, I could not map the tools to the carriages.  T1 **always** gets assigned to the X2 carriage, no matter what code changes I attempted in these two files.

status_screen_DOGM.cpp
   I was having issues in two methods with 4 tools/extruders so just got around the display portion of this and we used the serial port or OctoPrint for tool temp monitoring and switching.

DrumClock, 2020-07-08:
Hi! @ElectroFlux-SS and @Ziyayildizz3 👍👍
I'm glad there are more of us who are thinking about this mode. 😊😊

InsanityAutomation, 2020-07-14:
As one of the ones who has been through nearly all the underlying IDEX code, there is no trivial method to do this currently. Im focused purely on projects that pay the bills right now so itll be quite awhile until this gets more attention. One of the big items is that active extruder is currently used everywhere for determining the X stepper to use and its position index. That will need to be broken up. And it needs to be done in a way that expands to Y for the quad carriage setup.

DrumClock, 2020-07-14:
@InsanityAutomation 
Thanks for the information.
Can you please see issue #17941 why is the gap 198 mm in DUPLICATION and MIRRORED modes?
Thank you very much.

thinkyhead, 2020-07-14:
> there is no trivial method to do this currently

This small change will allow for more extruders, where the odd ones are located on one of the X carriages and the even ones located on the other X-carriage:

```diff
- if (movement_extruder()) X2_DIR_WRITE(v); else X_DIR_WRITE(v); \
+ if (movement_extruder() & 1) X2_DIR_WRITE(v); else X_DIR_WRITE(v); \
```

DrumClock, 2020-07-18:
Hi, which file (and method) to try this change in?

DrumClock, 2020-07-19:
I modified line 360 in ... scr\module\stepper.cpp ... as follows:

`else if (movement_extruder () & 1) X2_DIR_WRITE (v); else X_DIR_WRITE (v); \
`

Replacement of T0 and T1 will be successful.
When attempting to replace the T2 or T3, the X2 carriage constantly strikes the X-MAX endstop sensor.

**Behaviour (pronterface terminal) :**

> G28
SENDING:G28
echo:Active Extruder: 0

**X1 = print position
X2 = parking position**

> T1
SENDING:T1
echo:Active Extruder: 1

**X1 = move to parking osition
X2 = move to print position**

>T2
SENDING:T2
echo:Active Extruder: 2

**X2 = move to parking osition 
X2 = attempt to move through the X-MAX endstop
X1 = still in parking position**

> T3
SENDING:T3
echo:Active Extruder: 3

**X2 = attempt to move through the X-MAX endstop
X1 = still in parking position**

> T0
SENDING:T0
echo:Active Extruder: 0

**X2 = attempt to move through the X-MAX endstop
X1 = move to print position**

DrumClock, 2020-08-19:
### Please be able to set T0, T1 to carriage X1 and T2, T3 to carriage X2

DrumClock, 2020-09-28:
Hi,
Is anyone solving this question or should I close it?
Looks like there's no interest in it and nothing's happening.

thisiskeithb, 2020-09-28:
Keep in mind that all the code here is contributed by unpaid volunteers who work on it in their free time.

I've added a Help Wanted label to hopefully attract attention of someone who has the hardware, interest, and ability to fix/implement the change.

DrumClock, 2020-11-09:
HI @thinkyhead @InsanityAutomation 
I have a printer ready with IDEX and if possible,
I would like to try modified Marlin modules for the intended project.

Setting tools on carriages:
X1 = T0, T1
X2 = T2, T3

I'm very happy to help with testing, but I can't modify the program.

DrumClock, 2021-03-04:
> What you want to do isn't very much code. But it is complicated enough that I don't want to attempt anything without having a machine to code against and test the changes. The good news is, what you [...]

Hi @Roxy-3D
What about the division of tools into IDEX carriages?
I've been watching for a long time since nothing happened.

On my project, I tried it with the help of the classic IDEX for 2 extruders and a modified G-code for printing .... it's 100% functional.

If Marlin could define X1 = T0 / T1 and X2 = T2 / T3, I wouldn't have to modify the g-cod from the slicer at all .....

DrumClock, 2021-03-29:
**Because I probably won't see the adjustment of FW Dual X Carriage tool settings ....**
 I'm trying to solve it postscript G-code from Slicer

### **first cold test:**  https://www.youtube.com/watch?v=ED-3Wqm5rFQ

"Shame on you, Marlin programmers"

thisiskeithb, 2021-03-29:
> "Shame on you, Marlin programmers"

Stop demanding that people do the work for free or very little money.

Also, please stop creating issues in your repos and tagging users from this project asking for help. You’ve already been blocked by several users (including myself) because of this.

If someone wants to help you out or finds the work interesting, they will be in touch.

eightheads, 2021-03-29:
@DrumClock you do realize this is a **Feature Request** and no anything immediately necessary to the Marlin project or the user base as a whole right? If you feel the need to get this added quickly then put up some money to the Marlin devs, donate to the project, or learn some coding yourself. The Marlin devs are basically working for free to maintain Marlin for the printing community as a whole.

looxonline, 2021-03-29:
> **Because I probably won't see the adjustment of FW Dual X Carriage tool settings ....**
> I'm trying to solve it postscript G-code from Slicer
> 
> ### **first cold test:** https://www.youtube.com/watch?v=ED-3Wqm5rFQ
> "Shame on you, Marlin programmers"

Dude. Shame on you for being so entitled. Go learn how to code in c++, implement the functionality yourself and then submit a PR.

closed 2021-03-29 by DrumClock

DrumClock, 2021-03-30:
Sami jste omezení, 
několikrát jsem nabízel pomoc při testování - nikdo o to nestál.
Nabízel jsem také peníze - nikdo o to nestál.
Sám jsem investoval spoustu času a peněz do stavby tiskárny a vyladění V-extruderu v  tomto projektu.
Proč nevyužít tento systém když to HW umožňuje ?
Jakmile to nepoužívá 90% tiskařů a nejsou z toho peníze tak na to kašlete.

Končím , Marlin má spoustu chyb v IDEX modech a stejně to nikdo přes rok neřeší.

looxonline, 2021-03-30:
> Sami jste omezení,
> několikrát jsem nabízel pomoc při testování - nikdo o to nestál.
> Nabízel jsem také peníze - nikdo o to nestál.
> Sám jsem investoval spoustu času a peněz do stavby tiskárny a [...]

Please don't let the door hit you on the way out.

DrumClock, 2021-03-30:
> Please don't let the door hit you on the way out.

Sami mě vyháníte ..... je to vaše smůla a chyba

looxonline, 2021-03-30:
> Sami mě vyháníte ..... je to vaše smůla a chyba

I guess the intonation was lost in translation so let me make it clearer. People like you are a drain on the Marlin project. While it is nice to have people requesting features which helps developers identify possible areas for expansion, it is not nice to have people demanding features regardless of the project. This is especially true when the development work is being done for free, in people's spare time.

With that in mind, your attitude makes you *unwanted*. If you think your leaving is in any way a punishment to anyone else then you are sorely misguided. However, if you were willing to help more instead of just throwing demands around and shaming the developers then you may actually have been missed.

thinkyhead, 2021-03-30:
@DrumClock

I've been very busy with other issues, keeping up with Pull Requests, and trying to prepare a new release, so I haven't personally looked at the backlog of Feature Requests for quite a long time.

And I don't know if you've noticed, but there's a goram _pandemic_ underway.

### How does Open Source work?

It is very important to have the right perspective on what this repository is.

In spite of the illusory appearance, there is no **project team** waiting around to **do things for end-users** and that is not who you are writing to. You are writing to the world. You see, this repository is the result of a **collaboration** which is **open to everyone** on the planet, and people contribute mostly **for fun** and **in their spare time**. (Makerbase, BigTreeTech, et. al. excepted.)

My particular job is to try to cobble all these contributions together, keep all the files updated, keep the RepRap wiki up to date, and (eventually) expand and update the documentation, and I very rarely have time to actually implement features or extend existing features these days.

### What does "Request" mean?

You can think of a "feature request" as being exactly like a "help wanted" advertisement. If you need your fence painted, you can put out a "help wanted" advertisement, but maybe no one answers. You have some hope or need, but there is no one available to work on the problem! You can try writing an angry letter to the editor as the very next step, but maybe you just need to put up a **bigger ad**, or put it **somewhere else**, or offer some kind of **payment** … or maybe you just **paint your own fence**, or **get a whole new fence**!

Perhaps _RepRapFirmware_ or _Klipper_ can do what you want. You should bang on their door and demand some answers.

Certainly I do want to see this particular feature request implemented eventually, and probably I will end up implementing it myself. But _right at this moment_ I am very busy … typing this long comment.

If something is very much needed, then you must go out and do more real footwork and find help to get it done and not just put in a request —or demand— that _someone else_ —_anyone else_— invest all their experience, time, research and coding effort, and calories in you, when you may not even be desirable as an acquaintance.

Anyway, hopefully you now better understand how Open Source works.