MarlinFirmware/Marlin #18430: [BUG] (PAUSE - Skr E3 mini 1.2)

spooky79, 2020-06-26:
Hi

The pause function does not work. 
After pressing, nothing happens, after 15-20 seconds the main screen appears and the menu is locked
please help because I want to start long prints (with pause)
thank you !

[platformio.txt](https://github.com/MarlinFirmware/Marlin/files/4837195/platformio.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4837196/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4837197/Configuration_adv.txt)

ellensp, 2020-06-26:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

qwewer0, 2020-06-26:
Duplicate? (https://github.com/MarlinFirmware/Marlin/issues/18191)

spooky79, 2020-06-28:
on bugfix pause works but powerloss recovery does not work, after turning off the power and turning on the menu it is like after starting the printer

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4842500/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4842501/Configuration_adv.txt)
[platformio.txt](https://github.com/MarlinFirmware/Marlin/files/4842502/platformio.txt)

closed 2020-06-28 by spooky79