MarlinFirmware/Marlin #18432: Allow Cura to overwrite flow rate

gv2020, 2020-06-26:
With the most recent Marlin 2.x it is not possible to let cura control the flow rate. 

As a 3d printer owner using Cura 4.6
When i adjust the flow rate in Cura, create a gcode and send it to my printer
Then it is reflected on my printer
Current behaviour: when flow rate of an item is set to e.g 80% then printer ignores this and keeps flowrate at 100% unless i adjust it manually
Expected behaviour: when flow rate of an item is set to e.g. 80% then this is reflect on the printer and there is no need for me to adjust it manually on the printer

ManuelMcLure, 2020-06-26:
The flow rate specified by Marlin is a multiplier to the flow rate specified in Cura. So if you use 80% in Cura with 100% in Marlin you will get the 80% you want. If you then set the Marlin flow rate to 80% you will get the equivalent of 64% flow rate. So these are different values.

gv2020, 2020-06-26:
Thanks for the update. in that case, i am not under the impression cura is sending the right value as i see no difference when setting flow rate in cura. Closing this request. I will double check and file a bug where appropriate

closed 2020-06-26 by gv2020