MarlinFirmware/Marlin #18439: TM2209 - Sensorless Homing Z issue (ENDER3) SKR V1.4 Turbo

destruct123, 2020-06-27:
Bug Description

Sensorless homing works for X and Y, but simply dont trigger in Z axis.

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4839838/config.zip)

### Steps to Reproduce

1. Open Pronterface and conect to the printer
2. Home Z axis

**Expected behavior:** [What you expect to happen]

Stop at Z0

**Actual behavior:** [What actually happens]

Keeps trying to go down.

thisiskeithb, 2020-06-27:
> Keeps trying to go down.

Sensorless probing is not recommend for machines that use a leadscrew for this very reason.

https://github.com/MarlinFirmware/Marlin/blob/8474050adbe1bd4d363397cfedefaee996ac1e98/Marlin/Configuration.h#L951-L957

ABLomas, 2020-06-27:
But what's interesting - i have disabled sensorless homing on Z-axis, enabled bltouch (comm working - i can control it from menu), merged configs from earlier version (using winmerge for only changes i need) - still only X and Y homing is working. When i try to home Z - it tries to drill trough bed.
Tested on todays bugfix, re-checking configs, but so far it looks that Z-axis is ignored (SKR 1.4 turbo)

thisiskeithb, 2020-06-27:
Did you cut/remove the diag pin on Z? You’ll have to do that when using a probe.

ABLomas, 2020-06-27:
No, i did not and it was working fine on earlier build (approx 05.20 bugfix branch, can tell exact date later). AFAIK no need to cut anything with SKR 1.4 and TMC2209...

thisiskeithb, 2020-06-27:
You’ll need to cut/remove the diag pins from any axis with traditional endstops, probes, or filament runout (E0) since BigTreeTech did away with the diag jumpers.

ABLomas, 2020-06-27:
We are talking about some different things.
I did not connected any switches on X and Y axis to board and did not cut any pins. X and Y homing works now and was working earlier.
I connected bltouch to board and Z-axis homing was working fine on earlier firmware version, i printed many small prints with it and i'm sure it was working as expected without any pins cut.
What i did - merged configs and updated to new (todays) bugfix and now Z-axis homing does not work anymore.
Maybe firmware changed in a way that cutting pins is now mandatory. 

SKR manual say, that i need to cut switch only if i'm using mechanical endstops and not using stallguard. I do not use mechanical endstops and use stallguard on XY axis (this is where it is working) and do not use stallguard on Z axis and no mechanical endstop, but probe on Z axis.

(also, i'm not OP so maybe there are different issues here)

thisiskeithb, 2020-06-27:
> (also, i'm not OP so maybe there are different issues here)

I didn't see that, so most likely.

destruct123, 2020-06-27:
> > Keeps trying to go down.
> 
> Sensorless probing is not recommend for machines that use a leadscrew for this very reason.
> 
> https://github.com/MarlinFirmware/Marlin/blob/8474050adbe1bd4d363397cfedefaee996ac1e98/Marlin/Configuration.h#L951-L957

I think i'll keep my Z-axis sensor and keep sensorless other axis. Because even with 255 set in stallguard it still don't trigger my Z-axis sensorless homing. I'm not using a probe, just leting the axis hit it's end as the other axis, and i saw that it can damage the leadscrew if you let it keep going.

boelle, 2020-06-27:
Sensorless on screws will never work, i tried it even with lower amps on the steppers they are simply have to much torque for sensorless to work

deltaford, 2020-06-28:
@boelle  It absolutely can work but depends on having a machine designed for it.

I use it with a custom built CoreXY with SKR v.1.4 TURBO and sensorless TMC2209 on X, Y & Z and it works perfectly.
Z is designed with moving bed with 4 lead screws i corners, belt driven by a single stepper.

HunterWare, 2020-06-30:
Like boelle, I have tried z with leadscrew on all sensitivities and was never able to find one that worked at all, even unreliably.

You can certainly run a setup with the SKR1.4 + BLTouch without clipping the TMC2209 pins.  On my Ender5 with SKR1.4T and BLTouch I run sensorless on X/Y, and then wired up the BLTouch as documented.

You put the two plugs into "SERVOS" and "PROBE" (you might have to switch a pin or two in the header, check the wiring).

The you set the following:
- define USE_ZMIN_PLUG
- define Z_MIN_PROBE_ENDSTOP_INVERTING true
- define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
- define USE_PROBE_FOR_Z_HOMING
- define Z_MIN_PROBE_PIN P0_10
- define BLTOUCH
- define NOZZLE_TO_PROBE_OFFSET { -44, -8, -1.85 } // This is for my Ender5 with Creality mount
- #define Z_MIN_PROBE_REPEATABILITY_TEST // Do the M48 test to get confidence in probe

Now you are running sensorless X/Y so you don't have to clip them.  You are running BLTouch for Z stop, and the firmware will always deploy the probe for homes and thus you don't have to clip Z since the board is using a different pin.

Side note: this seems more like a request for help than a bug report / issue.  As in, the firmware is functioning as designed but is being misconfigured.
- Sensorless is known (and documented) to not work with leadscrews
- normal endstops are known not to work with SKR1.4 + unclipped TMC2209
- Probes work fine if you use a different pin, or clip the Z TMC2209 pin and use that GPIO.
- None of this is a firmware issue...

Maybe the issue should be closed?

HTH

closed 2020-06-30 by thisiskeithb