MarlinFirmware/Marlin #18441: Problems compiling marlin

Eternknight, 2020-06-27:
hello good morning I can not compile marlin with VSC before I could do it without problems I try to compile it for an ender 3 with a mks robin e3. this is the error that shows me, thank you very much,

Errors
PIO Core Call Error: "Processing mks_robin_e3 (platform: ststm32; board: genericSTM32F103RC; framework: arduino)\r\n--------------------------------------------------------------------------------\r\n\r\n\n\n========================= [FAILED] Took 13.96 seconds =========================\r\nEnvironment Status Duration\r\n---------------------------- -------- ------------\r\nmegaatmega2560 IGNORED\r\nmegaatmega1280 IGNORED\r\nrambo IGNORED\r\nFYSETC_F6_13 IGNORED\r\nsanguino_atmega644p IGNORED\r\nsanguino_atmega1284p IGNORED\r\nmelzi IGNORED\r\nmelzi_optiboot IGNORED\r\nat90usb1286_cdc IGNORED\r\nat90usb1286_dfu IGNORED\r\nDUE IGNORED\r\nDUE_USB IGNORED\r\nDUE_debug IGNORED\r\nLPC1768 IGNORED\r\nLPC1769 IGNORED\r\nSTM32F103RC IGNORED\r\nSTM32F103RC_fysetc IGNORED\r\nSTM32F103RC_bigtree IGNORED\r\nSTM32F103RC_bigtree_USB IGNORED\r\nSTM32F103RC_bigtree_512K IGNORED\r\nSTM32F103RC_bigtree_512K_USB IGNORED\r\nSTM32F103RE IGNORED\r\nSTM32F103RE_bigtree IGNORED\r\nSTM32F103RE_bigtree_USB IGNORED\r\nSTM32F4 IGNORED\r\nSTM32F7 IGNORED\r\nARMED IGNORED\r\nSTM32F103VE_longer IGNORED\r\nmks_robin IGNORED\r\nmks_robin_pro IGNORED\r\nmks_robin_lite IGNORED\r\nmks_robin_lite3 IGNORED\r\nmks_robin_e3 FAILED 00:00:13.960\r\nmks_robin_e3d IGNORED\r\nmks_robin_mini IGNORED\r\nmks_robin_nano IGNORED\r\nmks_robin2 IGNORED\r\njgaurora_a5s_a1 IGNORED\r\nSTM32F103CB_malyan IGNORED\r\nchitu_f103 IGNORED\r\nFLYF407ZG IGNORED\r\nFYSETC_S6 IGNORED\r\nSTM32F407VE_black IGNORED\r\nBIGTREE_SKR_PRO IGNORED\r\nBIGTREE_BTT002 IGNORED\r\nteensy31 IGNORED\r\nteensy35 IGNORED\r\nesp32 IGNORED\r\nlinux_native IGNORED\r\nSAMD51_grandcentral_m4 IGNORED\r\nrumba32_f446ve IGNORED\r\nmks_rumba32 IGNORED\r\ninclude_tree IGNORED\r\n==================== 1 failed, 0 succeeded in 00:00:13.960 ===================="

Compiling .pio\build\mks_robin_e3\src\src\HAL\shared\backtrace\unwarmbytab.cpp.o
In file included from Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/../HAL/./HAL_STM32F1/HAL.h:36,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/../HAL/./HAL_STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_e3\src\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./HAL_STM32F1/HAL.h:36,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\onboard_sd.cpp:15:
Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./HAL_STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_e3\src\src\HAL\HAL_STM32F1\onboard_sd.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./HAL_STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\HAL_spi_L6470.cpp:28:
Marlin\src\HAL\shared\../../inc/../HAL/./HAL_STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_e3\src\src\HAL\shared\HAL_spi_L6470.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\backtrace\../../../core/../inc/../HAL/./HAL_STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\backtrace\../../../core/../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\backtrace\../../../core/../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\backtrace\../../../core/serial.h:24,
                 from Marlin\src\HAL\shared\backtrace\backtrace.cpp:29:
Marlin\src\HAL\shared\backtrace\../../../core/../inc/../HAL/./HAL_STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.

zouden, 2020-06-27:
This problem is caused by PlatformIO making stm32duino the default core. Marlin requires maple. The Marlin platformio.ini file needs to be updated to specify maple.

I fixed this on my machine by adding

    board_build.core = maple

To line 46 of platformio.ini. This makes maple the default for all board environments. I don't know if this breaks it for other environments so maybe you just want to add the line to the relevant environment for your board.

ellensp, 2020-06-27:
See issue 1: https://reprap.org/forum/read.php?415,874307

ellensp, 2020-06-27:
This is already fixed in bugfix and will be in next release.

closed 2020-06-27 by ellensp

sjasonsmith, 2020-12-01:
We are finally updating ststm32 for maple-based boards and specifying the core, once this PR is merged: #20325