MarlinFirmware/Marlin #18444: [FR] X/Y backlash compensation on CoreXY printers

VanessaE, 2020-06-27:
### Description

I have a well-built and well-tuned Hypercube (if I do say so myself :smile: ) and backlash is the only other thing that could stand some improvement.

It isn't severe enough to distort vertical surfaces, holes, etc.  It's just enough that I have to over-extrude very slightly to make top surfaces watertight -- otherwise there are tiny gaps between pairs of infill lines (that is, two overlapping slightly, then a gap, then two more, and so on, classic signs of mild backlash).

Backlash compensation is available on normal cartesian machines like bedflinger and Ultimaker style, but is for some reason disabled on CoreXY.  Of course the movement/kinematics math is somewhat different on a CoreXY, but wouldn't the underlying algorithm still work if it weren't explicitly disabled?

kursatu, 2020-07-06:
This is something I certainly need to have. I also have a CoreXY printer, (Vking).
The surface quality is super important for my prints. The printer is used to print real products.
There is nothing else I can do to compensate for it mechanically.
The top layer, line gap/overlap is there, especially with layer heights of 0.1mm

palladin82, 2021-04-02:
Maybe there need to do motor based backlash compensation on direction change?
This will solve the problem of backlash on other types of mechanics as well.

palladin82, 2021-04-15:
Test needed [https://github.com/MarlinFirmware/Marlin/pull/21612](url)

thisiskeithb, 2022-02-25:
Added in #21612

closed 2022-02-25 by thisiskeithb