MarlinFirmware/Marlin #18449: [BUG] [Damages hardware] G28 Z0 ignores Z endstop and smashing into the bed

Uinden, 2020-06-27:
### Bug Description

My Z probe endstop damaged and needs to be replaced. Here is how.
You need Z probe that attaches manually using magnets. So probably most of user are safe.

### My Configurations

My config based on [Anet A6](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Anet/A6).
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4840317/config.zip)
I have `#define FIX_MOUNTED_PROBE` and huge 50mm tall self-made Z probe that attaches below nozzle using magnets.
It connected in parallel with Z endstop `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`. 

### Steps to Reproduce

1. Power on printer
2. Move Z to 60mm to attach 50mm tall probe
3. Send `G28 Z0`

**Expected behavior:** 
Move down until Z endstop triggered

**Actual behavior:** 
Nozzle moves down, Z endstop completely ignored, you can click it manually multiple times and it will not stop. Only emergency reset will save you.

#### Additional Information

I think I know where to search this bug.
If you power on printer and send `G28 Z0` printer will move 5mm up (not sure which config parameter is responsible for this) and then display message on screen `Home XY First` **(Why move at all???)**
**BUT!** If you 

- Power on printer
- Move to Z60 to attach probe
- Send `G28 Z0`

**IT WILL MOVE TO Z=5!!! Not Z+5!!!** Smashing and breaking everything it can on its way completely ignoring Z endstop. 
And after that it will display `Home XY First` message as a nice touch.

*If you power on printer, move to Z=60 and send `G28 X0 Y0` then it will move Z+5 as it should, not Z=5 as `G28 Z0` does.
So is there 2 bugs? 
One is that `G28 Z0` first moves and then says `Home XY First`
And second is that `G28 Z0` moves to Z=5 instead of Z+5 as `G28 X0 Y0` does.

sjasonsmith, 2020-06-30:
I think I have seen this before. Not with the exact setup you are using, but I've seen it move the nozzle toward the bed if I have jogged it up prior to homing. I didn't have a probe I was attaching to the bottom of the nozzle, so at the time it was a mere annoyance and not damaging my machine.

GMagician, 2020-06-30:
> One is that G28 Z0 first moves and then says Home XY First

I already noticed that. Raise is done in G28 function while error is raised in home when Z has to to this action, this is why you see it

GMagician, 2020-06-30:
Before homing this is how much Z is raised: `Z_HOMING_HEIGHT`
When Z pos is known it must go to Z_HOMING_HEIGHT otherwise current + Z_HOMING_HEIGHT but it should never go below current position

```c++
      if (destination.z > current_position.z) {
        if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPAIR("Raise Z (before homing) to ", destination.z);
        do_blocking_move_to_z(destination.z);
      }
```

When `Z_HOMING_HEIGHT` is not defined it will default to `Z_CLEARANCE_BETWEEN_PROBES` and this is why you have 5mm

To prevent this you may set Z_HOMING_HEIGHT to 0 or enable UNKNOWN_Z_NO_RAISE.

You may also enable debugging, m111 S255, home, and report log to have more info

GMagician, 2020-06-30:
oh other thing...are you usign bugfix? better to use it

sjasonsmith, 2020-07-01:
The code causing this is probably related to another issue which was closed by the submitter without being fixed: #16395. It would be good to take both scenarios into consideration when addressing this.

GMagician, 2020-07-01:
@sjasonsmith the code I posted above, if homing X or Y or safe Z home and Z is at known position will really lift Z everytime

GMagician, 2020-07-01:
But in such scenario I don't know if there is a solution. Since we don't know where Z is and we want to be sure to move away Z...we need to lift it (maybe adding a static variable that is set once Z has been lifted and skip lift if set may solve, but are we sure we don't put Z in danger in some situations?)

Uinden, 2020-07-02:
_This post is not related to current issue_
@GMagician 

> You may also enable debugging, m111 S255, home, and report log to have more info

> oh other thing...are you usign bugfix? better to use it

There is another problem now. Bugfix killing my bootloader (Marlin 2.0.5.3 works fine). I flash bugfix and I have empty screen and can't upload firmware anymore 

```
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x65
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x63
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x68
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6f
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x56
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x37
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x36
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x73
```

Luckily I have USBasp and I can restore [bootloader](https://raw.githubusercontent.com/Lauszus/Sanguino/master/bootloaders/optiboot/optiboot_atmega1284p.hex)
But I killed my bootloader like 5 times now trying to install bugfix branch.
Same for debug. I remove `#define LCD_BED_LEVELING` (because I need space) and enable `#define DEBUG_LEVELING_FEATURE` and after that I have empty screen and dead bootloader (stk500_getsync() messages) until I restore bootloader.

Looks like my cheap atmega1284p based board is not very good for this kind of tests.

Uinden, 2020-07-02:
> To prevent this you may set Z_HOMING_HEIGHT to 0

It does not raise from 0, but it does lower to 5.

> or enable UNKNOWN_Z_NO_RAISE.

Same problem. Does not raise from 0, but lowers from 6+ to 5

boelle, 2020-07-02:
what if you just send G28 Z and skip the 0 ?

Uinden, 2020-07-02:
@boelle 
`G28 Z` lowers from Z10 to Z5 too.
But I will say again, `G28 X0 Y0` raises to Z+5 from any position. Problem with `G28 Z0` that moving to Z=5 instead of Z+5.
`G28 X0 Y0` will move from `Z60` to `Z65`
`G28 Z0` will move from `Z60` to `Z5` smashing attached sensor.

GMagician, 2020-07-02:
@Uinden please re-read my previous post. I stated:

>  When Z_HOMING_HEIGHT is not defined it will default to Z_CLEARANCE_BETWEEN_PROBES and this is why you have 5mm

GMagician, 2020-07-02:
@Uinden 

> It does not raise from 0, but it does lower to 5.

setting such parameter to 0 prevent the z raise, so it problem is code I think you may solve it

GMagician, 2020-07-02:
> There is another problem now. Bugfix killing my bootloader (Marlin 2.0.5.3 works fine

That's not good...debugging could help to see where it moves Z to bad position

GMagician, 2020-07-02:
@Uinden I just read a post from @ellensp, not sure if he wrote it after reading this or not, but this may answer to your "bootcode erasing"

Uinden, 2020-07-02:
Sorry everyone, looks like it was fixed in bugfix
After I repaired my bootloader thanks to #18501 I was able to flash bugfix and when you `G28 Z0` it automatically homes `X+Y` first.
Here is leveling logs from both version. 
(I use octoprint, it sends some his data. I sent only `M111 S255` / `G1 Z10 F200` / `G28 Z0`)
[non_bugfix.txt](https://github.com/MarlinFirmware/Marlin/files/4865840/non_bugfix.txt)
[bugfix.txt](https://github.com/MarlinFirmware/Marlin/files/4865841/bugfix.txt)
So in current version
```
Send: G1 Z10 F200
Send: G28 Z0
Recv: echo:Home XY first
Recv: >>> do_blocking_move_to X-3.00 Y-5.00 Z5.00
```
So from Z10 it moves down to Z5

And in bugfix
```
Send: G1 Z10 F200
Send: G28 Z0
Recv: Raise Z (before homing) to 15.00
Recv: >>> do_blocking_move_to X-3.00 Y-5.00 Z15.00
Recv: >>> homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: >>> homeaxis(Z)
```
So from Z10 moves up to Z15 and homes everything.

Did I just wasted everyone's time or is there something I'm missing?

GMagician, 2020-07-02:
> or is there something I'm missing?

The name gives you the answer.....bugfix  :-D

It's possible this has been fixed

boelle, 2020-07-02:
@Uinden if you no longer have the issue, then press the green close button below

but please double check first

Uinden, 2020-07-02:
@boelle 
There is still code in `/src/gcode/calibrate/G28.cpp` that @GMagician found
```C++
    if (z_homing_height && (doX || doY || ENABLED(Z_SAFE_HOMING))) {
      // Raise Z before homing any other axes and z is not already high enough (never lower z)
      destination.z = z_homing_height + (TEST(axis_known_position, Z_AXIS) ? 0.0f : current_position.z);
      if (destination.z > current_position.z) {
        if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPAIR("Raise Z (before homing) to ", destination.z);
        do_blocking_move_to_z(destination.z);
      }
    }
```
And code that shows `Home XY First`
```C++
    // Disallow Z homing if X or Y are unknown
    if (!TEST(axis_known_position, X_AXIS) || !TEST(axis_known_position, Y_AXIS)) {
      LCD_MESSAGEPGM(MSG_ERR_Z_HOMING);
      SERIAL_ECHO_MSG(STR_ERR_Z_HOMING_SER);
      return;
    }
```
Not sure if it is possible now to call this code anymore.
I think I found commit https://github.com/MarlinFirmware/Marlin/commit/5f9624f70b78fecd7f2f27318b01578bcd038a06 where it was fixed. So now if you call `G28 Z0` it will home XY first.
Anyway I can't reproduce bug so I will close issue.

closed 2020-07-02 by Uinden

sjasonsmith, 2020-07-02:
I’m glad that worked. Now that you pointed it o it I recall another user was having homing crashed after executing G34 which led to those changes. I think a portion of the homing routine was erroring due to X/Y not being homed, but it would continue moving the print head around anyway.