MarlinFirmware/Marlin #18450: Additional PREHEAT settings not working[BUG] (short description)

maxz1158, 2020-06-27:
### Bug Description

Additional preheat settings look to have been added in the latest bugfix release (looks like up to 5 per the preheat parameters section in configuration_store.cpp, line 2556 and in menu_configuration.h lines ~324 and ~414).

When an additional material is added to configuration.h, like PETG for example, the firmware will fail to compile.

### My Configurations
[configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/4840903/configuration.files.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Add material to configuration.h
#define PREHEAT_3_LABEL       "PETG"
#define PREHEAT_3_TEMP_HOTEND 225
#define PREHEAT_3_TEMP_BED    70
#define PREHEAT_3_FAN_SPEED     0 // Value from 0 to 255
2. Compile
3. Errors and fails to compile

**Expected behavior:** [What you expect to happen]
Compile and upload to board with new preheat selection for PETG in the LCD menu.
**Actual behavior:** [What actually happens]
Fails to compile

#### Additional Information
I have this working with the following changes, but they are only to add one more preheat material setting.
There should be a global way to implement, but i haven't worked through that yet.
**conditionals_post.h**
line 2096 change from #define PREHEAT_COUNT 2 to #define PREHEAT_COUNT 3
This is where I think there should be a way to have the software count the number of materials specified in configuration.h and update the number accordingly.

**menu_configuration.h**
looks like the count was started from one instead of zero for the new adds.
starting at ~line 327
 from:  void menu_preheat_material3_settings() { _menu_configuration_preheat_settings(3); }
 to:      void menu_preheat_material3_settings() { _menu_configuration_preheat_settings(2); }
    #if PREHEAT_COUNT >= 4
from:      void menu_preheat_material4_settings() { _menu_configuration_preheat_settings(4); }
to:          void menu_preheat_material4_settings() { _menu_configuration_preheat_settings(3); }
      #if PREHEAT_COUNT >= 5
from:       void menu_preheat_material5_settings() { _menu_configuration_preheat_settings(5); }
to:           void menu_preheat_material5_settings() { _menu_configuration_preheat_settings(4); }

**language_en.h**
at line 91add labels for the number of preheat materials in configuration.h. This should probably go to 5 to match the number of preheat settings in menu_configuration.h.
 PROGMEM Language_Str MSG_PREHEAT_3                       = _UxGT("Preheat ") PREHEAT_3_LABEL;
 PROGMEM Language_Str MSG_PREHEAT_3_H                     = _UxGT("Preheat ") PREHEAT_3_LABEL " ~";
 PROGMEM Language_Str MSG_PREHEAT_3_END                   = _UxGT("Preheat ") PREHEAT_3_LABEL _UxGT(" End");
 PROGMEM Language_Str MSG_PREHEAT_3_END_E                 = _UxGT("Preheat ") PREHEAT_3_LABEL _UxGT(" End ~");
 PROGMEM Language_Str MSG_PREHEAT_3_ALL                   = _UxGT("Preheat ") PREHEAT_3_LABEL _UxGT(" All");
 PROGMEM Language_Str MSG_PREHEAT_3_BEDONLY               = _UxGT("Preheat ") PREHEAT_3_LABEL _UxGT(" Bed");
 PROGMEM Language_Str MSG_PREHEAT_3_SETTINGS              = _UxGT("Preheat ") PREHEAT_3_LABEL _UxGT(" Conf");

I'm sure there is a more elegant solution without having to manually make these changes, and I'm sure someone will be able to figure it out quicker than i can!

GMagician, 2020-07-01:
@maxz1158 please test PR

thisiskeithb, 2020-07-11:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/18488.

closed 2020-07-11 by thisiskeithb