MarlinFirmware/Marlin #18452: G29 Fails after 1st Probe

ykim22153, 2020-06-27:
Ender 3 Pro SKR Mini e3 V2.0 is failing after power cycling the system after firmware installation.  

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/4841100/Archive.zip)

Compile vanilla build (configuration files attached) for SKR Mini e3 v2.0.  
Install firmware.bin using SD Card on Mobo (install card, Power cycle printer)
Turn Printer off
Turn Printer on
Preheat bed to 60 degrees
Auto Home
Run ABL
Fails after first Probe completes and the probe/head move back up.

bill-orange, 2020-06-27:
I have dealt with this numerous times on several machines.  You are likely trying to probe outside the build plate area.  When this happens, you generally fail on the first probe.

This is just a sample from my Tronxy X3A.  You will likely need to so something like it.  This is only a sample from my configuration.h.  I would not expect this to work for you as-is. It will take a little trial and error,

```
// Tronxy X3A specific offsets
//#define TRONXY_X_BED_OFFSET -17
//#define TRONXY_Y_BED_OFFSET -25

//WEW 05/22/2020
#define TRONXY_X_BED_OFFSET -22
#define TRONXY_Y_BED_OFFSET -30

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS TRONXY_X_BED_OFFSET
#define Y_MIN_POS TRONXY_Y_BED_OFFSET
#define Z_MIN_POS 0
//#define X_MAX_POS X_BED_SIZE - (2 * TRONXY_X_BED_OFFSET)
//#define Y_MAX_POS Y_BED_SIZE - (TRONXY_Y_BED_OFFSET + 5)

//WEW 05/22/2020 
#define X_MAX_POS X_BED_SIZE - (2 * TRONXY_X_BED_OFFSET) - 25
#define Y_MAX_POS Y_BED_SIZE - (TRONXY_Y_BED_OFFSET + 5) - 25
#define Z_MAX_POS 300
  
```
If you post you problem in the Facebook Marlin or Ender 3 support group, someone will probably have something that will work for you already tested.  

The Facebook groups are the right place for this sort of question anyway.

ykim22153, 2020-06-29:
it's probing within the build plate area.  this seems to be an issue with the eeprom from what I am seeing.

boelle, 2020-07-02:
@ykim22153 and you remember to send M502 then M500 after you flash new firmware?

thinkyhead, 2020-07-03:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

ykim22153, 2020-07-04:
> @ykim22153 and you remember to send M502 then M500 after you flash new firmware?

yes.  I did.  I'll try to get a log up here.

ykim22153, 2020-07-05:
UPDATE:  Tried bugfix from 7/4/2020.  So far so good.  I have re-cycled the printer 7 times without any issues.

closed 2020-07-05 by ykim22153