MarlinFirmware/Marlin #18454: LCD 2004 with buttons doesn't work on SKR 1.4

AdamStastny, 2020-06-27:
### LCD 2004 with buttons doesn't work on SKR 1.4

### My Configurations

[config_marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4841373/config_marlin.zip)

### Steps to Reproduce

1. I normally instaled Marlin on my new SKR 1.4 board.
2. I tried to run it, but display blinks 2 times after start and than stays black.
3. Through repetier host printer is normally functioning (moving with axis).
4. I checked my old Marlin (but 1.0.x) configuration with display (back than I was  using original Anet mainboard) and it same, soo I think something is bad with pin definition maybe?

### Info about printer

I have Anet A8 stock display - LCD 2004 with 5 buttons, stock bed and extruder, motors

thinkyhead, 2020-06-28:
It appears that the pins have not been defined yet for this board to support character LCDs. But it shouldn't be too hard to figure out how to map them….

ellensp, 2020-06-28:
Not quite that easy 
It looks like a standard ZONESTAR_LCD 

using pins  (from diagrams only, may be flipped) 
5v    gnd
D4   N/C
D5   RS
D6   EN
D7   ADC

and exp1 on skr 1.4 is 
```
            5V | 1 2 | GND
 (LCD_D7) 1.23 | 3 4 | 1.22 (LCD_D6)
 (LCD_D5) 1.21 | 5 6   1.20 (LCD_D4)
 (LCD_RS) 1.19 | 7 8 | 1.18 (LCD_EN)
(BTN_ENC) 0.28 | 9 10| 1.30 (BEEPER)
```

The ADC pin on a skr 1.4 would be 1.30 which is analog capable as P1_30_A4 

BUT

This display is 5v powered and the ADC pin is not 5v tolerant.
With no buttons depressed this will damage the ADC with to much voltage.

thisiskeithb, 2020-06-29:
> With no buttons depressed this will damage the ADC with to much voltage.

The Anet/Zonestar full graphic controller is the same way. There’s a warning in the pins file to help prevent damage to the LCD/board:
https://github.com/MarlinFirmware/Marlin/blob/ca4ade4c44e50b65a588b041735ed852b71ed721/Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_3.h#L219-L231

ellensp, 2020-06-29:
Needs more than just a plug rewiring. Which is giving me pause, asking is this a good idea to add to standard pins?

AdamStastny, 2020-06-29:
Thank you for help. I guess easier and safer will be buing new lcd screen. Maybe I will try to run the display, but at Thursday because I am not at home now.

sjasonsmith, 2020-06-30:
@ellensp & @thisiskeithb it sounds like that check should move up into the common file.

Is this ever a viable LCD to use on any LPC/STM32 controller, or is it only an issue due to the exact pinout of the SKR boards? If it is almost never valid, maybe the check should just move moved's into the HAL's sanity checks or Conditional_LCD.h?

ellensp, 2020-06-30:
@sjasonsmith For the LPC1768/9 Its always an issue. All analog inputs are not 5v tolerant. and the LCD connectors provide 5v to the display so the resistor network for the switches generates voltages based on 5v.  All voltage values generated are greater than expected values and some greater than the allowed 3.3v.

I don't know STM32 that well, but google tells me they have the same issue. a-ref is 3.3v and not 5v tolerant.

thinkyhead, 2020-06-30:
If someone designs an adapter so that older 5V displays can be used with various boards, then we can support those pinouts. But the adapter design has to be posted someplace like Thingiverse, GitHub, Amazon, Gearbest….

boelle, 2020-07-02:
@thinkyhead @ellensp @sjasonsmith @thisiskeithb so is this an hardware issue and not marlin?

ellensp, 2020-07-04:
Yes, incompatible hardware.

closed 2020-07-04 by ellensp