MarlinFirmware/Marlin #18456: [BUG]  Sometime M400 is ignored, the next command is executed too soon

ChipCE, 2020-06-28:
### Bug Description

I have a short start Gcode to move Y to -1 then lower the Z. I inserted M400 between Y and Z move to make sure the Z only lower after the Y reach -1.  But sometime the Z still get lower when the bed mving to Y-1. 

Start code
<code>
M211 S0 ; Disable software endtop to allow nozzel move to y=-1
G1 Y-1 F2500.0 ; go outside print area
M400 ;
G1 Z0.3 F420.0 ; Lower Z
M400 ;
G92 E0.0 ; reset extruder distance position
G1 X190.0 E9.0 F1000.0 ; intro line
G1 X150.0 E21.5 F1000.0 ; intro line
</code>

### My Configurations
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4841691/config.zip)

### Steps to Reproduce

1. Slice the model, start gcode is insterted by slicer.
2. Print. -> M400 does not work
3. Reprint using the same  Gcode -> M400 work as expected.

thinkyhead, 2020-06-28:
What happens if you put `M420 S0` at the start of your g-code? Marlin won't combine a separate move only in Z with a move that is only in Y. Those will be two separate planner blocks run serially.

ChipCE, 2020-06-29:
@thinkyhead  Thank you! I will try and report back the result.

ChipCE, 2020-07-04:
@thinkyhead the <code>M420 S0</code> does not help. I don't known why it combine 2 line of gcode, and it not occur all the time, the occur rate is about 50%.

This is the gcode generated by the Prusa slicer
<pre>
; generated by PrusaSlicer 2.1.1+linux64 on 2020-07-04 at 07:36:23 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M201 X1000 Y1000 Z200 E6000 ; sets maximum accelerations, mm/sec^2
M203 X100 Y100 Z10 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E1.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
; ------------------- START GCODE -------------------
M862.3 P "CUSTOM" ; printer model check
G90 ; use absolute coordinates
M83 ; extruder relative mode

M117 Parking extruder
G28 ; Home
G1 X0 Y100 F2500 ; XY parking
M400 ;
G1 Z50 F420 ; Z parking
M400 ;

; then preheat (PRUSA SLICER ONLY)
M104 S240 ; set extruder temp
M140 S85 ; set bed temp
M190 S85 ; wait for bed temp
M109 S240 ; wait for extruder temp

M400 ; wait for complete
M300 S440 P500 ; play sound
M0 S30 Wait for clean nozzle ; wait for user to clean nozzle

;testing
M420 S0

G28 ; home
G29 ; mesh bed leveling

M211 S0 ; Disable software endtop to allow nozzel move to y=-1
G0 Y-1 F2500.0 ; go outside print area
M400 ;
G0 Z0.3 F420.0 ; Lower Z
M400 ;
G92 E0.0 ; reset extruder distance position
G1 X190.0 E9.0 F1000.0 ; intro line
G1 X150.0 E21.5 F1000.0 ; intro line
M400 ;
G1 Z2.0 F420 ; lift the extruder a bit
M400 ;
G1 Y0 ; move to Y0
M211 S1 ; re-enable software endstop
G92 E0.0 ; reset extruder distance position

G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M900 K0.046 ; Linear advance
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2

G1 E-1.40000 F2100.00000
G1 Z0.200 F10800.000
;AFTER_LAYER_CHANGE
;0.2
G1 Z0.400
G1 X104.184 Y66.569
G1 Z0.200
...
</pre>

thinkyhead, 2020-07-05:
How does `G4 P100` work? Is there any noticeable change?

ChipCE, 2020-07-10:
@thinkyhead the G4 seem to fix the problem.

closed 2020-07-10 by ChipCE