MarlinFirmware/Marlin #18457: [BUG] (short description)LCD_BED_LEVELING requires a programmable LCD controller error msg when trying to enable LCD_BED_LEVELING on CR3 v2

lreid7030, 2020-06-28:
### Bug Description

 I am getting an LCD_BED_LEVELING requires a programmable LCD controller error msg when I try to upload for a CR3 v2 with mesh bed leveling and LCD_BED_LEVELING options set.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-06-28:
Please attach your configuration files.

lreid7030, 2020-06-28:
Here they are.  Sorry about that...

> Please attach your configuration files.

[marlin_configs.zip](https://github.com/MarlinFirmware/Marlin/files/4841903/marlin_configs.zip)

lreid7030, 2020-06-28:
Thanks for the message.

They are attached now.

Cheers

ellensp, 2020-06-30:
This is not an error. 
You have a DWIN_CREALITY_LCD, this LCD is smart, It gets told what to display when via simple serial commands. But it has limits you cannot directly control what the pixels are doing.  
The  LCD_BED_LEVELING  feature requires a lcd that you can control individual pixels. 
This is simply not a option you can have with this display.

closed 2020-06-30 by ellensp