MarlinFirmware/Marlin #18458: UBL -Probe gets inactive

ircmgr, 2020-06-28:
There is an issue with UBL enabled. When i am doing a mesh level ( from LCD ) or a create cold mesh ( pla / abs "hot" mesh has the same behaviour ) it starts probing and after a while probe gets inactive and nozzle hits the bed hard.

Things tried :

* change the GRID_MAX_POINTS_X ( tried with values 10 ,4 ,5 )
* changed NOZZLE_TO_PROBE_OFFSET (tried 0,0,0 and the accurate ones)
* Z_PROBE_LOW_POINT ( from 0 to -7 , not going further it ll break my structure) 

Expected behavior : probe all points and give the mesh 
Actual behavior : Starts probing and after a while the probe gets inactive , probing continues slamming the bed but probe is inactive , even by putting a metal piece under it when high enough nothing triggers. 

After resetting probe is active again. Normal homing procedure runs as expected and prints start . the failure is while trying to configure UBL 

configuration.h 
   
```
/**
    * Marlin 3D Printer Firmware
    * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020006

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
[... 2231 lines cut ...]
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

      // Allow servo angle to be edited and saved to EEPROM
      //#define EDITABLE_SERVO_ANGLES

` 
```

boelle, 2020-07-02:
please attach the 2 config files in a zip file instead and remove the long config file text

just as the template told you to

ircmgr, 2020-07-02:
> please attach the 2 config files in a zip file instead and remove the long config file text
> 
> just as the template told you to

done

shitcreek, 2020-07-02:
Did it return an error message? What exactly do you mean 'after a while'? Like after a couple probe points? Does it fail at the same probe point each time?
Have you checked your cables and connections?
Disable the motors (**M18**) and move the print head around and see if if you can trigger the probe manually from all areas of the bed - if it fails then it maybe be a kinked/loosed wire.

ircmgr, 2020-07-03:
No it doesn't return an error. It is random point where it stops but always above point 14 in a 10x10 grid (100 points). If i reset on the last position an put my finger under the probe it will trigger so no cable faults.
If i manually move to any place it triggers as well. 

When this happens it ll start probing it ll reach a probe point number ( random above 14) and it ll keep probing there . Every time z goes below the low point it ll increase the probe point number and log the low point value in the mesh. But the head wont move! So if the issue starts at point 20 (assume somewhere close to center) then all points 21-100 will be probed at the same x/y  and in the end all these will have the low point value

thinkyhead, 2020-07-03:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

closed 2020-08-09 by github-actions[bot]