MarlinFirmware/Marlin #18461: [BUG] Cancel Object Numbers on statusscreen not corresponding to menu

swissnorp, 2020-06-28:
On the statusscreen it sais "Printing Object X" and in the Menu for Canceling there is "Object X-1".
Sorry for this stupid issue :). I am new to github. Could someone fix this, please?

 but in _cancel_object.ccp_ it says:
>  #if HAS_DISPLAY
>    if (active_object >= 0)
>      ui.status_printf_P(0, PSTR(S_FMT " %i"), GET_TEXT(MSG_PRINTING_OBJECT), int(active_object **+ 1**));
>    else
>      ui.reset_status();
>  #endif
>}

It would be an easy fix just to remove the "+1" ... what i did for compiling marlin.

sjasonsmith, 2020-06-30:
@swissnorp, you are welcome to submit a Pull Request to fix problems you find in Marlin.

If you aren't just going to post that PR yourself, I encourage you to go back and update your description to match the issue template you saw when filing the report. That consistent formatting will help make it easier for others to volunteer to help. Most important is the attaching of your configuration files.

GMagician, 2020-06-30:
This is the old problem of 0 based and 1 based values.
Objects in gcode are 0 based, but "humans" likes 1 based things.
Starting from idea to keep 1 on display, solution will be to subtract 1 to choosen item.
May be this the correct solution? I can't answer

boelle, 2020-07-27:
@swissnorp seen what the others wrote?

swissnorp, 2020-07-27:
I have seen the answers, yes. But i was on a vacation for some time.
I am new to github and not quite shure how to properly do a pull request with the files and the issue linked.

closed 2020-08-04 by thinkyhead