MarlinFirmware/Marlin #18463: [BUG] (power loss recovery not working on SKR E3 Mini  V1.2)

spooky79, 2020-06-28:
on Marlin bugfix (last version)  powerloss recovery does not work, after turning off the power and turning on the menu it is like after starting the printer
Printer Ender 3 Skr E3 mini V1.2
[platformio.txt](https://github.com/MarlinFirmware/Marlin/files/4842505/platformio.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4842506/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4842507/Configuration_adv.txt)

pmjdebruijn, 2020-06-29:
My guess would be:

`#define PLR_ENABLED_DEFAULT   false // Power Loss Recovery enabled by default. (Set with 'M413 Sn' & M500)`

spooky79, 2020-06-29:
was changed from the menu

sjasonsmith, 2020-06-30:
Duplicate of #17378.

@spooky79 as far as I can tell, this is the same issue as the issue linked above. Please continue any discussion in that report, so that anyone working on a fix can find all relevant information in one place.

In the future, please be sure to follow the template provided when you are reporting a new issue. That will make it easier for volunteers to understand and work on your issue.

closed 2020-06-30 by sjasonsmith