MarlinFirmware/Marlin #18469: [BUG] Cant compile if LCD is enabled

KrX3D, 2020-06-29:
### Bug Description

I downloaded the newest version of marlin from today (Marlin-bugfix-2.0.x).
When i do my changes in the Config file but dont enable my LCD i can compile it. But when i uncomment "REPRAP_DISCOUNT_SMART_CONTROLLER" i get this error:

`C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu_configuration.cpp: In function 'void _menu_configuration_preheat_settings(uint8_t)':
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu_configuration.cpp:310:52: error: 'struct preheat_t' has no member named 'fan_speed'
     editable.uint8 = uint8_t(ui.material_preset[m].fan_speed);
                                                    ^~~~~~~~~
In file included from C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu_configuration.cpp:31:0:
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu_configuration.cpp: In lambda function:
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu_configuration.cpp:311:117: error: 'struct preheat_t' has no member named 'fan_speed'
     EDIT_ITEM_N(percent, m, MSG_FAN_SPEED, &editable.uint8, 0, 255, []{ ui.material_preset[MenuItemBase::itemIndex].fan_speed = editable.uint8; });
                                                                                                                     ^
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu.h:409:39: note: in definition of macro '_MENU_INNER_P'
     MenuItem_##TYPE::action(plabel, ##V);                     \
                                       ^
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu.h:466:52: note: in expansion of macro '_MENU_ITEM_N_P'
 #define MENU_ITEM_N_P(TYPE, N, PLABEL, V...)       _MENU_ITEM_N_P(TYPE, N, false, PLABEL, ##V)
                                                    ^~~~~~~~~~~~~~
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu.h:492:53: note: in expansion of macro 'MENU_ITEM_N_P'
 #define EDIT_ITEM_N_P(TYPE, N, PLABEL, V...)        MENU_ITEM_N_P(TYPE, N, PLABEL, ##V)
                                                     ^~~~~~~~~~~~~
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu.h:493:53: note: in expansion of macro 'EDIT_ITEM_N_P'
 #define EDIT_ITEM_N(TYPE, N, LABEL, V...)           EDIT_ITEM_N_P(TYPE, N, GET_TEXT(LABEL), ##V)
                                                     ^~~~~~~~~~~~~
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu_configuration.cpp:311:5: note: in expansion of macro 'EDIT_ITEM_N'
     EDIT_ITEM_N(percent, m, MSG_FAN_SPEED, &editable.uint8, 0, 255, []{ ui.material_preset[MenuItemBase::itemIndex].fan_speed = editable.uint8; });
     ^~~~~~~~~~~
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu_configuration.cpp: In lambda function:
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu_configuration.cpp:311:117: error: 'struct preheat_t' has no member named 'fan_speed'
     EDIT_ITEM_N(percent, m, MSG_FAN_SPEED, &editable.uint8, 0, 255, []{ ui.material_preset[MenuItemBase::itemIndex].fan_speed = editable.uint8; });
                                                                                                                     ^
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu.h:414:58: note: in definition of macro '_MENU_INNER_P'
       (encoderLine == _thisItemNr, _lcdLineNr, plabel, ##V);  \
                                                          ^
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu.h:466:52: note: in expansion of macro '_MENU_ITEM_N_P'
 #define MENU_ITEM_N_P(TYPE, N, PLABEL, V...)       _MENU_ITEM_N_P(TYPE, N, false, PLABEL, ##V)
                                                    ^~~~~~~~~~~~~~
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu.h:492:53: note: in expansion of macro 'MENU_ITEM_N_P'
 #define EDIT_ITEM_N_P(TYPE, N, PLABEL, V...)        MENU_ITEM_N_P(TYPE, N, PLABEL, ##V)
                                                     ^~~~~~~~~~~~~
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu.h:493:53: note: in expansion of macro 'EDIT_ITEM_N_P'
 #define EDIT_ITEM_N(TYPE, N, LABEL, V...)           EDIT_ITEM_N_P(TYPE, N, GET_TEXT(LABEL), ##V)
                                                     ^~~~~~~~~~~~~
C:\Users\KrX\AppData\Local\Temp\arduino_build_942262\sketch\src\lcd\menu\menu_configuration.cpp:311:5: note: in expansion of macro 'EDIT_ITEM_N'
     EDIT_ITEM_N(percent, m, MSG_FAN_SPEED, &editable.uint8, 0, 255, []{ ui.material_preset[MenuItemBase::itemIndex].fan_speed = editable.uint8; });
     ^~~~~~~~~~~
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Users\\KrX\\AppData\\Local\\Arduino15\\packages\\Sanguino\\hardware\\avr\\1.0.3\\variants\\sanguino" "-IC:\\Program Files (x86)\\Arduino\\libraries\\LiquidCrystal\\src" "C:\\Users\\KrX\\AppData\\Local\\Temp\\arduino_build_942262\\sketch\\src\\lcd\\menu\\menu_game.cpp" -o "C:\\Users\\KrX\\AppData\\Local\\Temp\\arduino_build_942262\\sketch\\src\\lcd\\menu\\menu_game.cpp.o"
Bibliothek LiquidCrystal in Version 1.0.7 im Ordner: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Sanguino.
`

it works fine with the stable version of marlin.

My Board is a sanguinololu 1284 (not the P version) 16Mhz

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4845418/Marlin.zip).

### Steps to Reproduce

1. Select Sanguino 1284 with 16Mhz in Arduino IDE or PlattformIO
2. enable REPRAP_DISCOUNT_SMART_CONTROLLER

**Expected behavior:** i expect it to compile successfully ;) 

**Actual behavior:** Gettting error shown above when compiling

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-06-29:
In Marlin/src/lcd/ultralcd.h is is 
```
  typedef struct {
    TERN_(HAS_HOTEND,     uint16_t hotend_temp);
    TERN_(HAS_HEATED_BED, uint16_t bed_temp   );
    TERN_(HAS_FAN,        uint16_t fan_speed  );
  } preheat_t;
```
fan_speed is conditional on HAS_FAN

but the test HAS_FAN is not in bugfix 

you can just add #define HAS_FAN 1  to your Configuration.h to get it to compile.

ellensp, 2020-06-29:
If you replace the following in menu_configuration.cpp 
```
      editable.uint8 = uint8_t(ui.material_preset[m].fan_speed);
      EDIT_ITEM_N(percent, m, MSG_FAN_SPEED, &editable.uint8, 0, 255, []{ ui.material_preset[MenuItemBase::itemIndex].fan_speed = editable.uint8; });
```
with this
```
    #if HAS_FAN
      editable.uint8 = uint8_t(ui.material_preset[m].fan_speed);
      EDIT_ITEM_N(percent, m, MSG_FAN_SPEED, &editable.uint8, 0, 255, []{ ui.material_preset[MenuItemBase::itemIndex].fan_speed = editable.uint8; });
    #endif 
```
It compiles, but I haven't had a chance to test beyond compiling.

KrX3D, 2020-06-29:
hi,

ok that worked, my display is also working.

About what fan di you speak, the hotend fan? i think its directly connected to 12v/5v and never got a problem.

so will this fix be added in the next version of marlin, do i dont need to think about changing it each time ;)

ellensp, 2020-06-29:
On a old boards, like this, it doesn't have software controlled fans. But the lcd code assumed that there was a software controlled fan. Thus it failed to compile.   Fana directly on power don't count as firmware can't see or do anything with them.

ellensp, 2020-06-29:
It will get added to bugfix, as soon as I (or someone else) gets a chance to do so.  Just leave issue open for now thanks.

ellensp, 2020-07-01:
Fix has been applied to bugfix. Closing issue.

closed 2020-07-01 by ellensp