MarlinFirmware/Marlin #18472: [BUG] Unable to home 2nd axis with auto home

fuganater, 2020-06-30:
### Bug Description

I'm working on a CoreXY printer and using an SKR 1.3 with TMC2209's with sensorless homing. The issue is when I hit AUTO HOME, the printer will home X, moves 2 or 3mm on Y, then homes Z. If I enable "HOME_Y_BEFORE_X", then Y homes fine and X does not. This is a known good version of Marlin working on another printer and these features work. 

### My Configurations

[BLV MNG Cube Config.zip](https://github.com/MarlinFirmware/Marlin/files/4849143/BLV.MNG.Cube.Config.zip)

### Steps to Reproduce

AUTO HOME, the printer will home X, moves 2 or 3mm on Y, then homes Z. If I enable "HOME_Y_BEFORE_X", then Y homes fine and X does not.

**Expected behavior:** I expect Auto Home to work properly

**Actual behavior:** Auto home only homes the 1st axis

sjasonsmith, 2020-06-30:
Does it work if you home each axis separately using `G28 X` and `G28 Y`?

This is probably a duplicate of another issue. The answer to that question is important to know for sure. I can try to help consolidate these once I’m back at a computer.

fuganater, 2020-06-30:
Yes. I can home X and Y via pronterface and the LCD individually with no issue. It is only an issue with Auto home

sjasonsmith, 2020-06-30:
Duplicate of #16420

That issue has 2130 in the title, but the overall symptom seems to be the same. Several people have echoed the same failure (and workaround!), so I suspect that CoreXY/Sensorless is the important detail, and not the exact drivers used.

sjasonsmith, 2020-06-30:
@thinkyhead did you mean to re-open this? I closed it as a duplicate as we had several issues tracking the same CoreXY Sensorless Homing issue.

thinkyhead, 2020-06-30:
Testing the "unlock on reopen" action. Seems to work!

closed 2020-06-30 by thinkyhead

sjasonsmith, 2020-07-06:
Duplicate of #16420