MarlinFirmware/Marlin #18476: [BUG] TFT_LITTLE_VGL_UI external font loading, special characters no displayed, missing screens

thisiskeithb, 2020-06-30:
### Bug Description

This is a continuation of discussion from PR https://github.com/MarlinFirmware/Marlin/pull/18438. Several issues still exist:

- [ ] External fonts load, but do not display correctly
- [ ] Default LVGL 6.0 font does not display special characters
- [ ] Several screens are missing compared to MKS' old/proprietary TFT firmware
- [ ] Upgrade to LVGL 7.0
- [ ] Rename `TFT_LITTLE_VGL_UI` since it was renamed to LVGL - "Light and Versatile Graphics Library" in 7.0.
- [ ] I'll add more as this issue is discussed

### My Configurations

This is a general issue with MKS's LVGL implementation, so it applies to all configs with `TFT_LITTLE_VGL_UI` enabled.

makerbase-mks, 2020-07-01:
@rhapsodyv @thisiskeithb 
The bugfix-2.0.x version submitted by rhapsodyv has been tested and found that Chinese and Russian languages can be displayed. Due to the problem of address offset, the display is not complete enough. I need to change it.

The following picture shows the problem that occurred after I updated the latest font. I need to modify the font offset in the code to solve this problem.
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/master/Firmware/mks_font
[image]

The interface shown in the following picture is the normal interface displayed after I updated the old version of the font
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/v2.0.5.3/Firmware/mks_font
[image]

thisiskeithb, 2020-07-01:
> The bugfix-2.0.x version submitted by rhapsodyv has been tested and found that Chinese and Russian languages can be displayed. Due to the problem of address offset, the display is not complete enough. I need to change it.

Sounds like a simple fix. There's still an issue with some special characters not being displayed.

In your photos above, "franais" should be "français":
https://github.com/MarlinFirmware/Marlin/blob/e5bc9d31cc87be5c7ef2bd3cd70c2599f76d2bd7/Marlin/src/lcd/extui/lib/mks_ui/tft_multi_language.h#L535

and "espaol" should be "español":
https://github.com/MarlinFirmware/Marlin/blob/e5bc9d31cc87be5c7ef2bd3cd70c2599f76d2bd7/Marlin/src/lcd/extui/lib/mks_ui/tft_multi_language.h#L541

@rhapsodyv was able to update to LVGL 7.0 without modifying your fork. but I don't think they've tested special characters yet.

ftoz1, 2020-07-01:
Seem to me wrong way add lvlg issues to many files as y present. There should be one lvlg starter file instead of ultralcd.cpp which inicialize gui.

thisiskeithb, 2020-07-01:
> Seem to me wrong way add lvlg issues to many files as y present. There should be one lvlg starter file instead of ultralcd.cpp which inicialize gui.

Can you clarify what this means?

This is a beta feature that is a work in progress. Expect frequent updates as this feature is updated and expanded to more hardware.

ftoz1, 2020-07-01:
Its not depend if beta of what stage it is, but all lcd issues should be only in /lcd dir. And marlin.cpp should call init gui and update gui. There cant be gui issues across all marlin files. It sucks me as hell on marlin. The reason is simple if y want to use another ui/gui, y get big stop by many ui issues in gcodes and in hundred files.  I am now refresh work on it and seems to share some experimental working version till week. But i am always trouble with basic things on marlin. Dual Z motor not work on latest versions, so always need to solve basic things.

thisiskeithb, 2020-07-01:
It sounds like you have a misunderstanding of how Marlin is organized and configured and those are not related to LVGL/this issue.

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

ftoz1, 2020-07-01:
https://www.youtube.com/watch?v=3coW24_y9d8&feature=youtu.be

rhapsodyv, 2020-07-01:
> @rhapsodyv @thisiskeithb
> The bugfix-2.0.x version submitted by rhapsodyv has been tested and found that Chinese and Russian languages can be displayed. Due to the problem of address offset, the [...]

Did you fixed the offset? What’s to commit to my branch?

rhapsodyv, 2020-07-01:
> good luck.
> https://www.youtube.com/watch?v=3coW24_y9d8&feature=youtu.be

All the lvgl code is isolated in lcd folder. There’re only two from Marlin core: init and update loop.

ftoz1, 2020-07-01:
I am saying that, because original marlin cause many troubles by that. There are a lot of ui issues everywhere and i am stuck on it.  If want keep marlin with open mind changeable ui, its good to keep file structure.

GMagician, 2020-07-01:
@thisiskeithb and Italian is "Italiano" not "Italia" :-)

thisiskeithb, 2020-07-01:
> @thisiskeithb and Italian is "Italiano" not "Italia" :-)

I didn’t write the translations. I’m just pointing out the missing characters.

GMagician, 2020-07-01:
@thisiskeithb you may think about a new job :-D

thisiskeithb, 2020-07-01:
> @thisiskeithb you may think about a new job :-D

What do you mean by that?

GMagician, 2020-07-01:
translator

thisiskeithb, 2020-07-01:
Marlin is accepting pull requests, so feel free to correct translation errors.

makerbase-mks, 2020-07-02:
@thisiskeithb Thank you for your suggestion，I have submitted my changes.
I merged @rhapsodyv 's code, the test found that it worked very well.
The following content is my submission link
https://github.com/MarlinFirmware/Marlin/pull/18500

I haven't updated the graphics library to 7.0 yet, I think I will cooperate with @rhapsodyv  to do this job well.
The following are my tests on robin_nano_v1.2 and V2.0 after the merger
[image]
[image]

ftoz1, 2020-07-02:
How fast is approx draw page on nano with that?

rhapsodyv, 2020-09-03:
I think we can update the progress

thisiskeithb, 2020-10-05:
@makerbase-mks Is updating to LVGL 7.x still planned? [They are up to version 7.6 now](https://github.com/lvgl/lvgl).

ellensp, 2020-10-22:
several parts of the code for MKS_UI code calls "card."   without  checking sdsupport is actually enabled 
egs 
mks_ui/draw_print_file.cpp:549:3: error: 'card' was not declared in this scope
mks_ui/printer_operation.cpp:237:5: error: 'card' was not declared in this scope

rhapsodyv, 2020-10-22:
> several parts of the code for MKS_UI code calls "card." without checking sdsupport is actually enabled
> egs
> mks_ui/draw_print_file.cpp:549:3: error: 'card' was not declared in this scope
> mks_ui/printer_operation.cpp:237:5: error: 'card' was not declared in this scope

LVGL ui will work without SDSUPPORT because it needs to loads the assets.

Maybe we should put a sanity check to remember the user to enable it.

ellensp, 2020-10-22:
makes sense, but then the blocks of code like "#if ENABLED(SDSUPPORT)" don't....

rhapsodyv, 2020-10-22:
> makes sense, but then the blocks of code like "#if ENABLED(SDSUPPORT)" don't....

Sure... those blocks don't make sense... at least now.

Maybe in the future, when we have a light version of the lvgl, without assets loading

closed 2020-10-22 by thisiskeithb