MarlinFirmware/Marlin #18478: [BUG] ABL probing fails

fuganater, 2020-06-30:
### Bug Description

Auto Bed Level via the LCD and G29 via Pronterface fail after probe #3

### My Configurations

[BLV MNG Cube Config.zip](https://github.com/MarlinFirmware/Marlin/files/4854769/BLV.MNG.Cube.Config.zip)

### Steps to Reproduce

G29 - bed leveling fails after probe point #3 and says "probing failed"

**Expected behavior:** ABL to work

**Actual behavior:** Fails on probe #3

#### Additional Information

I'm using an SKR 1.3 with TMC 2209 drivers on a CoreXY printer. My Z sensor is a LJ12A3-4-Z connected with an optocoupler to Z min. I have a 5x5 grid and it probes each location 3 times. I'm running Marlin 2.0.5.3. I have this exact same setup on an i3 and it works as it should.

sjasonsmith, 2020-07-01:
Please try the following:
1. Re-test with the latest state of the `bugfix-2.0.x` branch, if you have not already. It is possible your issue has already been fixed.
2. Enable leveling debug prints with `M111 S32` (I see you already have DEBUG_LEVELING_FEATURE enabled in your Configuration.h)
3. Execute `G29` to reproduce the failure.
4. Post the debug output here.

That debug output may provide insight into the reason it is failing. You may only need to provide the last portion of the output where the failure occurs. If you have a lot of data to post, consider putting it on pastebin and linking to it from here.

sjasonsmith, 2020-07-01:
This is mostly a re-statement of what I said above, but I didn't provide the complete instructions which are now provided when debugging leveling/probing issues. Here it is for completeness:

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin bugfix-2.0.x to test with the latest code.
- Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send M502 and M500 to ensure your Configurations are applied.
- Issue the command M111 S247 to enable maximum logging.
- Perform a G28 to do your standard homing procedure.
- Do a G29 to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a .TXT file and attach it to your next reply.
- Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Dracrius, 2020-07-09:
@fuganater try disabling/ commenting out `#define ADAPTIVE_STEP_SMOOTHING` I spent the last day dialing prob issues to this setting in #18598 . I also had failures if I enabled `#define MULTIPLE_PROBING` and after checking your configs you have both settings enabled. After all my testing though I am sure you can leave `#define MULTIPLE_PROBING 3` though it may be unneeded after you disable `#define ADAPTIVE_STEP_SMOOTHING` as my Bltouch meshes are fairly consistent with the default single probe mode.

sjasonsmith, 2020-07-18:
This is not using a BLTouch, so it would not be related to the unstable PWM signal that `ADAPTIVE_STEP_SMOOTHING` can cause. I don't think there is anything actionable on this right now, until @fuganater replies with additional information.

thestephenmarshall, 2020-08-08:
This has been happening to me for months now on various branches I've tried and `ADAPTIVE_STEP_SMOOTHING` has been *disabled* the entire time until just the past week. API docs might be missing some instructions perhaps? `2.0.4.4` and earlier I can just `G29 P1` to generate a probed mesh (just as the docs say) then `G29 J` if want to tilt that mesh.

boelle, 2020-08-08:
@fuganater still with us? do you still have the problem or is it solved?

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands

fuganater, 2020-08-10:
> @fuganater still with us? do you still have the problem or is it solved?
> 
> Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands

Yes I am here! I did upgrade to the newest Bugfix and it seems to have solved the issue!!

thisiskeithb, 2020-08-10:
> I did upgrade to the newest Bugfix and it seems to have solved the issue!!

Great 🙂

closed 2020-08-10 by thisiskeithb