MarlinFirmware/Marlin #18482: [BUG] unable to build for chitu_v5_gpio_init on linux (case sensitivity)

andersking, 2020-07-01:
using platformio to compile for chitu_v5_gpio_init fails with missing "boards/boards.h" (on linux). 

The platformio maple variant is CHITU_F103 (in ~/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants)

but compilation expects "chitu_f103" - I managed to get it to compile by symlinking CHITU_F103 to chitu_f103. 

Not sure of the correct place to fix this though? (or whether it needs to be fixed in platformio instead).

(latest commit was 2d6c17b044fc14162727fc5933de57d94973dfd5 on bugfix-2.0.x)

ellensp, 2020-07-01:
Edit platformio.ini set board to uppercase ie
```
[env:chitu_f103]
platform      = ${common_stm32f1.platform}
extends       = common_stm32f1
board         = CHITU_F103
```
rename Marlin/buildroot/share/PlatformIO/boards/chitu_f103.json to CHITU_F103.json

andersking, 2020-07-01:
Works for me. 
Thanks.

closed 2020-07-11 by ellensp