MarlinFirmware/Marlin #18489: [FR] Add menu item to bed leveling menu to use G35

KyleLampkin, 2020-07-01:
It would be nice to have the G35 tramming feature added to the bed leveling menu so that you don't need to connect with a pc or write a gcode program.
# NO SUPPORT REQUESTS PLEASE

Support Requests posted here will be automatically closed!

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. See https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question.

For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org Marlin Forum http://forums.reprap.org/list.php?415
- Tom's 3D Forums https://discuss.toms3d.org/
- Facebook Group "Marlin Firmware" https://www.facebook.com/groups/1049718498464482/
- Facebook Group "Marlin Firmware for 3D Printers" https://www.facebook.com/groups/3Dtechtalk/
- Marlin Configuration https://www.youtube.com/results?search_query=marlin+configuration on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a Bug Report at https://github.com/MarlinFirmware/Marlin/issues/new/choose).

boelle, 2020-07-01:
could you please remove  half the post... i think when you read your own post you will get a smile on your face

Tannoo, 2020-07-03:
Add to all bed leveling menues or just ABL?

KyleLampkin, 2020-07-03:
I would think just the ABL and possibly UBL since you need a probe of some kind to make it work!

InsanityAutomation, 2020-07-22:
The menu item probably makes sense where the G34 menu is. I added this to my existing PR #18737 where I was adding another menu right next to it.

NetTecture, 2020-08-02:
I just was getting bugfix today and - I have a menu item, but it does nothing. Is that work in progress? Because - why have it in a LCD menu when I need to connect a terminal ANYWAY to see the output and get some use out of it. This shoudl be able to run either full  in menu, or - be removed from the menu, if one asks me.

thisiskeithb, 2020-11-07:
Implemented in https://github.com/MarlinFirmware/Marlin/pull/20000

closed 2020-11-07 by thisiskeithb