MarlinFirmware/Marlin #18492: Source code for CR10S Pro V2

Zs0lty87, 2020-07-01:
Please attach the source code for Creality CR10S Pro V2 as it is not within the examples. The hex file is working good but when I try to configure one from the main source it doesn't work, printer doesn't start up, only showing the startup screen.

The only thing I need to change is the probe offsets but I can't do it without the configured source code.

I'll be a sponsor if you upload the files.

Regards

thisiskeithb, 2020-07-01:
> Please attach the source code for Creality CR10S Pro V2 as it is not within the examples.

You’ll need to reach out to Creality and request the source code as we don’t have control over that.

Example configs hosted [here](https://github.com/MarlinFirmware/Configurations) are built and contributed by volunteers for the benefit of the community.

boelle, 2020-07-02:
closing this one as it would have to be provided by creality, nothing we can do here except what we allready done

closed 2020-07-02 by boelle

davidcranor, 2020-07-13:
Hi, i'm trying to find configuration for my 10S Pro V2, but I do not see a file in the example configs directory, I only see one for the 10s (non-pro non v2), is there somewhere else I should be looking?  thanks!

thinkyhead, 2020-07-13:
If you get the firmware from Creality3D, it should include the configurations.

ellensp, 2026-09-25:
https://www.creality.com/download/cr-10s-pro-v2-3d-printer 
The firmware source is hidden here. https://cdn.creality.com/ow/official/b42dae1d-10a6-42ba-b522-4f580c118c9e.zip
 
Attached is a copy
[CR-10S PRO V2.zip](https://github.com/user-attachments/files/32632682/CR-10S.PRO.V2.zip)