MarlinFirmware/Marlin #18494: G29 J Reporting different results when triggered manually vs during a print file

RFBomb, 2020-07-02:
### Bug Description

G29 J has wildly different results from when I initiate it via a print and when I trigger it manually. 

### My Configurations

Using UBL with BL-Touch. 
Ender 3. 
[Configs_7-1-2020.zip](https://github.com/MarlinFirmware/Marlin/files/4861628/Configs_7-1-2020.zip)

### Steps to Reproduce

G29 P0
G28
G29 J

**Expected behavior:** [What you expect to happen]

G29 J should be very close to identical between subsequent runs of itself. 
There should be no difference triggering it manually vs during a print as pre-printing gcode. 

**Actual behavior:** [What actually happens]

I'll let the pictures speak for themselves here (The results for manual trigger are about what is expected. I have utter contempt for the one scratching my bed though):

https://www.reddit.com/r/ender3/comments/hjodh9/bltouch_g29_j_inconsistent_manual_trigger_is/

[Cold Bed - 27.4 C - Triggered Manually](https://preview.redd.it/sq2sbn1oqc851.png?width=623&format=png&auto=webp&s=f5c27526512152883ac3a76f2177e489b81a1c20)

[80C Bed  after 30 minutes heating up - triggered manually ](https://preview.redd.it/8908kek0rc851.png?width=608&format=png&auto=webp&s=470f0b4a27120c6574ddc3cdcaa11d3dfa52daab)

[80C bed taken a few minutes after the one shown above - This one was done as part of the startup code in a print file](https://preview.redd.it/18oi5wx5rc851.png?width=668&format=png&auto=webp&s=fb2f0941e31c360ade7c3558000b25d6de99e896)

#### Additional Information

shitcreek, 2020-07-02:
Could you post your gcode file and the mesh results (eg **G29 J T**)?
I am not having any issue UBL. There are very subtle and expected differences between each bed leveling but overall they are the same for me:
```
Manual:
  0       1       2       3       4      
0 -0.275  -0.356  -0.437  -0.518  -0.599 
1 -0.180  -0.261  -0.342  -0.422  -0.503 
2 -0.084  -0.165  -0.246  -0.327  -0.408 
3 +0.012  -0.069  -0.150  -0.231  -0.312 
4 +0.107  +0.026 [-0.055] -0.136  -0.217 

Print from SD:
   0       1       2       3       4      
0  -0.272  -0.350  -0.428  -0.505  -0.583 
1  -0.181  -0.259  -0.336  -0.414  -0.492 
2  -0.090  -0.168  -0.245  -0.323  -0.401 
3  +0.001  -0.076  -0.154  -0.232  -0.310 
4 [+0.092] +0.015  -0.063  -0.141  -0.219  

Manual:
  0       1       2       3       4      
0 -0.252  -0.340  -0.428  -0.516  -0.604 
1 -0.160  -0.248  -0.336  -0.424  -0.512 
2 -0.069  -0.157  -0.244  -0.332  -0.420 
3 +0.023  -0.065  -0.153  -0.241  -0.328 
4 +0.115  +0.027 [-0.061] -0.149  -0.237 

Print from SD:
  0       1       2       3       4      
0 -0.276  -0.356  -0.435  -0.515  -0.595  
1 -0.182  -0.261  -0.341  -0.421  -0.501 
2 -0.087  -0.167  -0.247  -0.327  -0.406 
3 +0.007  -0.073  -0.153  -0.232  -0.312 
4 +0.101  +0.021 [-0.058] -0.138  -0.218
```
[image]
[image]

RFBomb, 2020-07-02:
Here is the GCODE I use in my print startup:

`;Generated with Cura_SteamEngine 4.6.1
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
M117 Preheating Bed + Extruder
M140 S50 ; Set Heat Bed temperature
M104 S180; Warm Extruder just below operating temp
M190 S50 ; Wait for Heat Bed temperature

G28 ; Home all axes
M117 Auto-Leveling Bed
;G29 P1 S0;	Auto bed-level (BL-Touch)
G29 L0 A ;Load and Activate the UBL mesh in Specified Slot
G29 J ; Tilt Mesh Bed Leveling

M117 Getting the extruder up to temp
M104 S205 ; Set Extruder temperature
M117 Begin Purge`

Once I saw it start to purge, I then stopped the printer, as I knew the issue would be present and my goal was purely to get the tilt. I have the G29 P1 S0 in there commented out so that way I can easily enable it if I want to, but typically don't use it.  I have tried it with both `G29 L0 A` enabled and with that command commented out. Since I have zero'd out my mesh and used M500 afterward, the results with and without it are the same. 

The problem is purely that the print for some reason has come out like in the picture I linked to.

I'm in process of cleaning my hot end and putting in a 0.2mm nozzle, so I'll post the G29T results once able.

RFBomb, 2020-07-03:
Here is the Bed Report:

Manual @ 50C

```
Recv: Bed Topography Report:
[...]
Recv:     (  5,220)                                                              (225,220)
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:  9 | +0.062  +0.050  +0.037  +0.025  +0.012  -0.000  -0.013  -0.026  -0.038  -0.051
Recv:    |
Recv:  8 | +0.067  +0.055  +0.042  +0.030  +0.017  +0.005  -0.008  -0.020  -0.033  -0.046
Recv:    |
Recv:  7 | +0.072  +0.060  +0.047  +0.035  +0.022  +0.010  -0.003  -0.015  -0.028  -0.040
Recv:    |
Recv:  6 | +0.077  +0.065  +0.052  +0.040  +0.027  +0.015  +0.002  -0.010  -0.023  -0.035
Recv:    |
Recv:  5 | +0.083  +0.070  +0.057  +0.045  +0.032  +0.020  +0.007  -0.005  -0.018  -0.030
Recv:    |
Recv:  4 | +0.088  +0.075  +0.063 [+0.050] +0.037  +0.025  +0.012  -0.000  -0.013  -0.025
Recv:    |
Recv:  3 | +0.093  +0.080  +0.068  +0.055  +0.043  +0.030  +0.017  +0.005  -0.008  -0.020
Recv:    |
Recv:  2 | +0.098  +0.085  +0.073  +0.060  +0.048  +0.035  +0.023  +0.010  -0.003  -0.015
Recv:    |
Recv:  1 | +0.103  +0.090  +0.078  +0.065  +0.053  +0.040  +0.028  +0.015  +0.003  -0.010
Recv:    |
Recv:  0 | +0.108  +0.095  +0.083  +0.070  +0.058  +0.045  +0.033  +0.020  +0.008  -0.005
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:     (  5,  5)                                                              (225,  5)
[...]
Recv: ok P31 B63
```

[image]

-------------------------------------------------

Here is the test GCODE file I used, and the results, run immediately afterward. 

[Tilt_Test.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/4867126/Tilt_Test.gcode.txt)
```
M117 G29 Tilt Test
M140 S50.0 ; Set Heat Bed temperature
M190 S50.0 ; Wait for Heat Bed temperature
G28 ; Home all axes
G29 P0
G29 A
G29 J ; Tilt Mesh Bed Leveling
G29 T

```

Results:
```
Send: N11 M117 [######### ]  96%*27
Recv: ok N11 P31 B63
Send: N12 G29 T*85
[...]
Recv: Bed Topography Report:
[...]
Recv:     (  5,220)                                                              (225,220)
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:  9 | +0.054  +0.041  +0.029  +0.017  +0.004  -0.008  -0.020  -0.033  -0.045  -0.058
Recv:    |
Recv:  8 | +0.060  +0.048  +0.035  +0.023  +0.011  -0.002  -0.014  -0.027  -0.039  -0.051
Recv:    |
Recv:  7 | +0.066  +0.054  +0.041  +0.029  +0.017  +0.004  -0.008  -0.020  -0.033  -0.045
Recv:    |
Recv:  6 | +0.072  +0.060  +0.048  +0.035  +0.023  +0.011  -0.002  -0.014  -0.027  -0.039
[... 106 lines cut ...]
Recv:  5 | -0.685  -0.969  -1.255  -1.555  -1.796  -2.028  -2.259  -2.534    .       .
Recv:    |
Recv:  4 | -0.782  -1.059  -1.333 [-1.626] -1.848  -2.069  -2.297  -2.561    .       .
Recv:    |
Recv:  3 | -0.814  -1.078  -1.368  -1.674  -1.898  -2.116  -2.358  -2.629    .       .
Recv:    |
Recv:  2 | -0.884  -1.154  -1.420  -1.711  -1.940  -2.146  -2.410  -2.673    .       .
Recv:    |
Recv:  1 | -0.973  -1.255  -1.500  -1.788  -2.022  -2.272  -2.518  -2.775    .       .
Recv:    |
Recv:  0 |   .       .       .       .       .       .       .       .       .       .
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:     (  5,  5)                                                              (225,  5)

```

[image]

```
[L<index>] | Load  a mesh. If no index is given, load the previously-activated mesh. The  given mesh index will be used for subsequent Load and Store operations.
```

I wouldn't think that loading a mesh (especially a ZERO'D mesh would cause an issue like this. But it would appear that when `G29 J` command is run, it automatically saves the mesh to the specified slot. I verified this by zero'ing a mesh, saving it to slot 0, running G29 J, the  loading the mesh with G29 L0. Performed a G29 T produced a tilted mesh very close to the first good one I posted in this discussion. 

I was able to get this horrid tilt to occur several times, but the more i tested the less it occurred. It was almost as if the tilt was occasionally saved as the mesh itself, so the next time it performed a tilt, it tilted a tilted mesh, resulting in  the horid amounts. The concering thing to me here though is that nothing was positive in the mesh when it faulted out. So I'm not really sure whats going on here.

closed 2020-08-08 by github-actions[bot]