MarlinFirmware/Marlin #18498: [BUG] Maybe bad TMC initialization?

GMagician, 2020-07-02:
I'm facing a weird situation that I'm trying to analize.

Cnf: TMC2209 with sensorless home

It's seems that if I power up system via USB and then turnon 24V power (and it seems sometimes, not always) when I start homing, x moves to home, reach endstop and then if stutter without ever terminating home.

If I turn on 24V and then connect usb, it seems this never happens.

Please note:
1) that threshold is "almost" calibrated and sometime X or Y start/stop immediately without doing correct home (then threshold is not for sure too low)
2) On shield there is a diode to power stepsticks vmot via 3.3V when 24V is not present

Is there some initialization/calibration TMC do when powered up then can't be fullfilled when missing 24V?

ellensp, 2020-07-02:
And how did you indicate to the firmware that you enabled 24v late and it needed to reinitialize?

boelle, 2020-07-02:
please use discord for discussions

GMagician, 2020-07-02:
@ellensp my suspect is that when vmot comes to TMC something happens to it, or that when it's powered with low voltage it doesn't initialize correctly....

closed 2020-07-02 by GMagician

GMagician, 2020-07-02:
@ellensp maybe I found how to replycate this, and it really seems related to missing 24V.

How to generate it:
Power 24V, connect USB:
  homing is ok
Remove 24V but keep usb:
  homing is bad
Don't know what happens but going on to investigate

GMagician, 2020-07-02:
Ok, m122 give an asnwer and there is a problem that I'm not sure it can be detected...after removing 24V and reapplyig 24V I see that vsense and msteps changes
from 0.11 to .325 and mstep from 16 to 8

GMagician, 2020-07-02:
Here is m122 before removing 24V (Z2 has wrong current, I just notices now)
15:05:37.387 : X	Y	Z	Z2	E
15:05:37.387 : Address		0	0	0	0	0
15:05:37.418 : Enabled		false	false	false	false	false
15:05:37.418 : Set current	900	900	900	1000	250
15:05:37.441 : RMS current	887	887	887	994	244
15:05:37.472 : MAX current	1251	1251	1251	1402	344
15:05:37.472 : Run current	28/31	28/31	28/31	17/31	7/31
15:05:37.472 : Hold current	14/31	14/31	14/31	8/31	3/31
15:05:37.503 : CS actual	14/31	14/31	14/31	8/31	3/31
15:05:37.525 : PWM scale	16	16	16	10	4
15:05:37.556 : vsense		1=.18	1=.18	1=.18	0=.325	1=.18
15:05:37.572 : stealthChop	true	true	true	true	true
15:05:37.603 : msteps		16	16	16	16	16
15:05:37.625 : tstep		max	max	max	max	max
15:05:37.625 : PWM thresh.
15:05:37.641 : [mm/s]
15:05:37.657 : OT prewarn	false	false	false	false	false
15:05:37.657 : triggered
15:05:37.672 : OTP		false	false	false	false	false
15:05:37.688 : off time	4	4	4	4	4
15:05:37.719 : blank time	24	24	24	24	24
15:05:37.719 : hysteresis
15:05:37.741 : -end		2	2	2	2	2
15:05:37.773 : -start		1	1	1	1	1
15:05:37.773 : Stallguard thrs	80	84	0	0	0
15:05:37.773 : DRVSTATUS	X	Y	Z	Z2	E
15:05:37.804 : sg_result	0	0	0	0	0
15:05:37.826 : stst
15:05:37.842 : olb
15:05:37.873 : ola
15:05:37.888 : s2gb
15:05:37.920 : s2ga
15:05:37.942 : otpw
15:05:37.957 : ot
15:05:37.989 : 157C
15:05:38.004 : 150C
15:05:38.026 : 143C
15:05:38.058 : 120C
15:05:38.058 : s2vsa
15:05:38.058 : s2vsb
15:05:38.058 : Driver registers:
15:05:38.073 : X	0xC0:0E:00:00
15:05:38.073 : Y	0xC0:0E:00:00
15:05:38.089 : Z	0xC0:0E:00:00
15:05:38.089 : Z2	0xC0:08:00:00
15:05:38.089 : E	0xC0:03:00:00
15:05:38.105 : Testing X connection... OK
15:05:38.105 : Testing Y connection... OK
15:05:38.127 : Testing Z connection... OK
15:05:38.127 : Testing Z2 connection... OK
15:05:38.142 : Testing E connection... OK

and this is after I removed and reapplyerd 24V (report only differences)

15:06:57.700 : RMS current	1602	1602	1602	994	441
15:06:57.738 : MAX current	2259	2259	2259	1402	622
15:06:57.769 : CS actual	31/31	31/31	31/31	31/31	31/31
15:06:57.784 : PWM scale	36	36	36	36	36
15:06:57.816 : vsense		0=.325	0=.325	0=.325	0=.325	0=.325
15:06:57.869 : msteps		8	8	8	8	8
15:06:57.954 : off time	3	3	3	3	3
15:06:57.985 : blank time	36	36	36	36	36
15:06:58.016 : -end		-3	-3	-3	-3	-3
15:06:58.038 : -start		6	6	6	6	6
15:06:58.038 : Stallguard thrs	80	84	0	0	0
15:06:58.333 : Driver registers:
15:06:58.339 : X	0xC0:1F:00:00
15:06:58.339 : Y	0xC0:1F:00:00
15:06:58.355 : Z	0xC0:1F:00:00
15:06:58.355 : Z2	0xC0:1F:00:00
15:06:58.370 : E	0xC0:1F:00:00

GMagician, 2020-07-02:
@ellensp I did another test. TMC reset when I remove 24V, not when I give it back...maybe is something related to my shield board....investigating
@boelle this may be a serious problem, if not related to my H/W, but solution will not be easy (not possible to know when 24V is missing)

GMagician, 2020-07-02:
@Bob-the-Kuhn sorry to bother you but you are the one with the best knowledge on tmc. What may be the reason of the reset registers reported above with relationship to 24V (vmot) removing? Thanks for your attention

Evg33, 2020-07-02:
M502
M501 
in startup code can help

GMagician, 2020-07-02:
@Evg33 keeping 5V on usb will not reset Cpu so data is not reloaded from eeprom (by the way I did M502 & m500 a tons and redone just now but nothing changes)

GMagician, 2020-07-02:
@Bob-the-Kuhn, @ellensp, @boelle & @thinkyhead  Hi everybody, sorry to spam this but I think it may be interesting to know.
When TMC2209 is powered from power supply (12V or 24V) and this power supply will miss for some reason and tmc will be powered with 3.3V (also on on vmot, and this is usually done to prevent firing it when usb powers chip but vmot is absent), this will generate this weird issue. Then using stepstick protectors (like watterott ones) will for sure save tmc but will give another issue.

I think it's important to know. I don't use discord because it's too dispersive. Hope this will not bother you, in such base please forgive me.

GMagician, 2020-07-02:
**ops I forgot...if 5V is sent when 24V is missing will solve this**

Evg33, 2020-07-02:
M502
M501
for reinit TMC

GMagician, 2020-07-02:
@Evg33 ok, now I understood why you said about M502 & M500 but I found a better solution and this solution may help everyone.

As stated before issue "born" when you switch off PSU supply and keeps usb (**note switch off with usb on. If you power with usb and then power PSU everything is ok**). This is because vmot on my board (or with stepstick protector) is powered by vcc that it's 3.3V. On my board I have diodes on 3.3V and 5V to act as "stepstick protector" in this case I'm able to send to vmot a 5V when psu is off. With this solution tmc dont' lose configuration and once PSU is on again I'm able to work without M50X

Don't know why tmc2209 act this way and if only 2209 are affected or not (I suspect also  2208)

Evg33, 2020-07-02:
All non standalone TMC with same problem.
2209,2208,2130,... etc

GMagician, 2020-07-02:
Than if you use usb 5V with a diode to vmot you solve this issue