MarlinFirmware/Marlin #18499: [BUG] (shorFysetc S6 TFT81050 С экрана в меню настроек вы не можете увеличить чувствительность без датчика, только уменьшить. t description)

Roman-ru, 2020-07-02:
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/4863774/Configuration.h.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4863775/Configuration_adv.zip)

### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-07-02:
Please add som description in english, follow the template

sjasonsmith, 2020-07-03:
Google translate translates the title as this:

(shorFysetc S6 TFT81050 From the screen in the settings menu, you cannot increase the sensitivity without a sensor, only decrease it. t description)

It is a duplicate of this one which was already closed due to missing information
#18497

@Roman-ru, it sounds like there might actually be an issue here. Thank you for providing configuration files. We still need more information to know if this is a bug.

1. Please edit the title to be in English
2. Update the description to fill in:
   - Steps to Reproduce
   - Expected Behavior
   - Actual Behavior

It is ok if you use Google Translate to provide English descriptions. Please let us know if you do this, though, since it can occasionally change the meaning of things for you.

Roman-ru, 2020-07-04:
https://cloud.mail.ru/stock/m97ncBhrYeH7J4wYSoXhwAgm  
 I made a video, maybe it'll help.

Roman-ru, 2020-07-04:
I'm an ordinary PC user, and I don't know English, so it's hard for me to understand what's needed. In advanced settings - sensorless Homing there are buttons - and q, so, when you press - the value decreases, everything is fine, but when you click on the q except for the sound of pressing nothing happens the value does not increase.

sjasonsmith, 2020-07-05:
Please test the change in this pull request. I believe it will fix your problem, but I do not have the hardware to test it.
https://github.com/MarlinFirmware/Marlin/pull/18554

sjasonsmith, 2020-07-06:
Closing, since @Roman-ru confirmed on a PR comment that the problem is now fixed.

closed 2020-07-06 by sjasonsmith