MarlinFirmware/Marlin #18501: [BUG] env:melzi and env:sanguino1284p flash size to large, leads to overwriting of bootloader.

ellensp, 2020-07-02:
### Bug Description

This was discussed way back in March in discord, but never got fixed. 

There are two common bootloader options on a sanguino_atmega1284p
The older bootloader is 4096 bytes in size leaving 126976 byes for firmware.
The newer optiboot bootloader is only 1024 bytes in size leaving 130048 bytes for firmware.

But platformio presumes the newer optiboot bootloader and always sets 130048 bytes for firmware size

So if you select env:melzi or env:sanguino1284p and upload the firmware, If it exceeds 126976 bytes it starts to overwriting the beginning of the larger bootloader.
 
The env: can set a new max firmware size with the following line.
board_upload.maximum_size = 126976   <= search for this on discord to find the discussion.

### My Configurations

Config files are not required in this rare instance. But will add a example shorty

### Steps to Reproduce

1. use a sanguino_atmega1284p with older bootler  (It wont upload with 115200 that optiboot uses) 
2. Compile marlin to a size > 126976 but less than 130048   (don't have an example handy... ill try to generate one later.)
3. Upload new firmware. 
4. Reboot controller and try to upload firmware again. you just get the following errors   

"avrdude: stk500_cmd(): programmer is out of sync" and "avrdude: stk500_cmd(): protocol error"

**Expected behavior:** 

No issues upload a second time.

**Actual behavior:**

You get "avrdude: stk500_cmd(): programmer is out of sync" and "avrdude: stk500_cmd(): protocol error" on attempt to upload second time.

#### Additional Information

search discord for "board_upload.maximum_size = 126976"

Uinden, 2020-07-02:
> The newer optiboot bootloader is only 1024 bytes in size leaving 130048 bytes for firmware.

Where to get it? I keep destroying my bootloader.
I use [7 years old](https://github.com/Lauszus/Sanguino/blob/master/bootloaders/optiboot/optiboot_atmega1284p.hex) one, and [optiboot repo](https://github.com/Optiboot/optiboot/tree/master/optiboot/bootloaders/optiboot) doesn't have 1284p .hex file. I see there is makefile but I am not sure I can do it.

ellensp, 2020-07-02:
Its parts of arduino sanguino package. https://github.com/Lauszus/Sanguino/tree/master/bootloaders/optiboot

Uinden, 2020-07-02:
@ellensp 
This is exactly what I use.
I upload it using USBasp and avrdudess (GUI for avrdude)
`-c usbasp -p m1284p -P COM1 -b 57600 -B 0.5 -e -U flash:w:"optiboot_atmega1284p.hex":a `
Then I connect board over USB and upload Marlin
```
Sketch uses 127746 bytes (98%) of program storage space. Maximum is 130048 bytes.
Global variables use 3656 bytes (22%) of dynamic memory, leaving 12728 bytes for local variables. Maximum is 16384 bytes.
```

```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/arch/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/home/arch/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 5.0
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: reading input file "/tmp/arduino_build_548734/Marlin.ino.hex"
avrdude: writing flash (127746 bytes):

Writing | ################################################## | 100% 17.39s

avrdude: 127746 bytes of flash written
avrdude: verifying flash memory against /tmp/arduino_build_548734/Marlin.ino.hex:
avrdude: load data flash data from input file /tmp/arduino_build_548734/Marlin.ino.hex:
avrdude: input file /tmp/arduino_build_548734/Marlin.ino.hex contains 127746 bytes
avrdude: reading on-chip flash data:

Reading | #############################avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avr_read(): error reading address 0x0000
    read operation not supported for memory "flash"
avrdude: failed to read all of flash memory, rc=-2
avrdude: stk500_recv(): programmer is not responding

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected
```

After that my board and bootloader is dead. Empty screen and errors when attempting to upload another Marlin.

```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
```

Need to attach USBasp and flash bootloader again.
After that I upload smaller Marlin

```
Sketch uses 125292 bytes (96%) of program storage space. Maximum is 130048 bytes.
Global variables use 3645 bytes (22%) of dynamic memory, leaving 12739 bytes for local variables. Maximum is 16384 bytes.
```

And it works.

ellensp, 2020-07-02:
usbasp is junk, doesn't decode address > 64k correctly. Will write crap all over the flash.

Uinden, 2020-07-02:
Finally. It works.
So I used **Arduino Uno** as **Arduino as ISP** and selected **Burn Bootloader** in **Arduino IDE**.
Uploaded 127746 bytes Marlin over USB twice and it works. Never worked before.

So somehow USBasp uploads "currupted"(?) bootloader that can't process 127746 byte sketch, but processes 125292 without any problem multiple times. This is some black magic I will not be able to understand.
Does bootloader has some code over 64k address that USBasp can't write and without that part at 64k address bootloader can't write to 127746+ address?
Anyway it works now. Thank you for your help!

closed 2020-07-04 by ellensp

bitboy85, 2020-08-24:
Thx for posting. I guess this exactly describes how i crashed my bootloader XD. I have a Creality 1.1.5 Silent Board for the Ender 3 which comes with a preinstalled bootloader. So it seems to use the large one.
If i understand right it is possible to use the small bootloader AND having a program storage space of 130048 bytes.