MarlinFirmware/Marlin #18505: [BUG] Compile error after updating to STM32 7.0.0

Miraculix200, 2020-07-02:
I'm not sure if this is a bug with Marlin or something else?

Board: MKS_ROBIN_NANO

After updating the STM32 platform on PlatformIO to ST STM32 7.0.0 > STM32F103VE using the command "platformio update" under Linux, I can't compile Marlin anymore. 

I've tried it with a lightly modified vanilla version of branch bugfix-2.0.x (set the correct board MKS_ROBIN_NANO and enabled FSMC_GRAPHICAL_DISPLAY).

```
Compilation terminates with this error:

In file included from Marlin/src/HAL/STM32F1/dogm/../../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin/src/HAL/STM32F1/dogm/../../../inc/../HAL/HAL.h:26,
                 from Marlin/src/HAL/STM32F1/dogm/../../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/dogm/u8g_com_stm32duino_fsmc.cpp:29:
Marlin/src/HAL/STM32F1/dogm/../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

Any idea how to fix it?

shitcreek, 2020-07-02:
Please try searching for the issue first. These are the threads of I've found that may help with you.
https://github.com/MarlinFirmware/Marlin/issues/18288#issuecomment-644915489
https://github.com/MarlinFirmware/Marlin/issues/18288
https://github.com/MarlinFirmware/Marlin/issues/18440
https://github.com/MarlinFirmware/Marlin/pull/18496

thisiskeithb, 2020-07-02:
I'm seeing the same issue on a Robin Nano build after PIO updated. It might be wise to lock down the STM32 version in PIO to <7.0 until it's fixed.

sjasonsmith, 2020-07-02:
My PR to upgrade to 7.0 (#18496) is only for boards using HAL/STM32.  In the bugfix-2.0.x branch boards using HAL/STM32F1 are already capped to ststm32 6.1.

thisiskeithb, 2020-07-02:
Not on the Nano apparently. Or PIO is ignoring the cap since it happily updated on my build.

sjasonsmith, 2020-07-02:
I'd call that a merge error.
That was probably added about the same time all the STM32F1 targets were refactored in the platformio.ini.

thisiskeithb, 2020-07-02:
Could be. I don't know why or _if_ they need to be different.

sjasonsmith, 2020-07-02:
These are the only unique lines:

`extra_scripts = buildroot/share/PlatformIO/scripts/mks_robin_nano35.py`
`MKS-LittlevGL=https://github.com/makerbase-mks/MKS-LittlevGL/archive/master.zip`

You could probably replace it with this and it would work fine:

```
#
# MKS Robin Nano (STM32F103VET6) - MKS UI (LVGL)
#
[env:mks_robin_nano35]
platform      = ${common_stm32f1.platform}
extends       = ${env:mks_robin_nano}
extra_scripts = buildroot/share/PlatformIO/scripts/mks_robin_nano35.py
lib_deps      = ${common_stm32f1.lib_deps}
     MKS-LittlevGL=https://github.com/makerbase-mks/MKS-LittlevGL/archive/master.zip
```

If you make that edit you should relocate it in the file to be next to the `mks_robin_nano` while you are at it.

thisiskeithb, 2020-07-02:
Using the updated `mks_robin_nano35` environment above, It fails instantly with the following despite having a board defined:

```prolog
> Executing task: platformio run <

Processing mks_robin_nano35 (platform: ststm32@<6.2.0; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: BoardConfig: Board is not defined
================================================ [FAILED] Took 0.57 seconds ================================================
```

Wiping out PIO folders (including a super delete/reset) didn't help, so maybe it has to use `platform = ststm32`?

sjasonsmith, 2020-07-02:
I think I got the `extends` line wrong, try:
`extends       = env:mks_robin_nano`

sjasonsmith, 2020-07-02:
I went back and edited that snippet a few replies up so anybody can copy/paste it and it will work now.

thisiskeithb, 2020-07-02:
I put in PR https://github.com/MarlinFirmware/Marlin/pull/18516 to get this fixed for everyone.

sjasonsmith, 2020-07-04:
@Miraculix200 this fix has been merged into the `bugfix-2.0.x` branch. Since the problem was so clear and easy to reproduce, I'm  assuming it is now solved for you as well. I'll close this issue, but please let us know if it doesn't work for you when you test with the updated branch.

closed 2020-07-04 by sjasonsmith