MarlinFirmware/Marlin #18506: [BUG] Idling after a G28; Probe crashes into bed when performing a G29

STaveras, 2020-07-02:
Idling after homing can cause the probe to crash into the bed when deployed to build a mesh of the build platform.

1. Home by sending G28
2. Idle after 5 minutes
3. Send a G29 P1 to Rebuild Mesh

Expected: Raise Z, Deploy Probe, Begin Probing
Actual: Deploy Probe, Begin Probing

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4866924/Configuration.zip)

sjasonsmith, 2020-07-03:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

There was an issue related to homing where sometimes Z homing could be skipped and cause crashes. That should be prevented in the latest `bugfix-2.0.x`. I think that was updated a couple weeks ago?

lilasiian8, 2020-07-08:
I am on the the bugfix-2.0.x branch; updated this on Sunday, The probe crashes into the bed on slow/second probe after I printed something and then the printer sat idle. My work around was to home (G28) and G29 P1 from the LCD; instead of terminal via OctoPrint.

Dracrius, 2020-07-14:
@STaveras Pull Request #18637 is more then likely the solution to your issues. 

The problem is made far worse with `ADAPTIVE_STEP_SMOOTHING` enabled as it is in your configs as `ADAPTIVE_STEP_SMOOTHING` currently can occupy 100% processor power for the stepper-interrupt alone by stepping at the wrong speed. This is not the main issue but another that makes things worse the pull request fixes the Servo Pin priority which prevents `ADAPTIVE_STEP_SMOOTHING` misbehavior from causing probing interference. A separate fix for `ADAPTIVE_STEP_SMOOTHING` is also in the works.

STaveras, 2020-07-15:
Thanks a lot. I was on commit 7b6629c0 when I submitted the issue. My apologies for not updating this issue earlier, but I haven't had a problem since 9ee891c4 as there was fix for TMC drivers and homing. I also removed the "UNKNOWN_Z_NO_RAISE" that was defined by default with the example config for the SKR Mini E3 V2.0. This latest fix is a godsend, as it gets rid of the massive discrepancies that would show up as huge dips or voids in the mesh. Thanks again.

closed 2020-07-15 by STaveras

thisiskeithb, 2020-07-15:
> I also removed the "UNKNOWN_Z_NO_RAISE" that was defined by default with the example config for the SKR Mini E3 V2.0

Odd. It's disabled in the ones we host here: https://github.com/MarlinFirmware/Configurations/blob/5224f4944f93694dc0506b13fe38a6a973f9c033/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%202.0/Configuration.h#L1101

thinkyhead, 2020-07-15:
Current list of configs with `UNKNOWN_Z_NO_RAISE`:

- Creality/CR-10mini/MEEB-3DP
- Creality/Ender-3/MEEB-3DP
- Einstart-S
- EXP3D/Imprimante multifonction
- FlashForge/Creator 2X
- FlashForge/CreatorPro
- Kingroon/KP3
- Qidi/Qidi 1
- Wanhao/Duplicator 4S