MarlinFirmware/Marlin #18517: [BUG] (short description)

SaschaKP, 2020-07-03:
### Bug Description
on Mightyboard RevE, with 2.0 bugfix (pulled at 12:30 UTC time, on 02/07/2020) the first heater stopped working, I searched in the code why this could happen, but didn't found anything that could give me that hint, just to avoid misunderstandings, the previous firmware (2.0.0.5, stable one) do work, but in that particular firmware there is a problem with homing axes using a probe, that in the recent one is fixed.

### My Configurations

[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/4867171/MarlinConfig.zip)

### Steps to Reproduce

pick up a Replicator1, as you can see in the config, apart from the sensor and different pulley, everything is as the machine requires to be usable, compile and upload, heater 0/A will stop working (the one in the right side)
With the previous firmware instead, it will work as normal.

ellensp, 2020-07-03:
enable PINS_DEBUGGING in Configuration_adv.h and send a M43, check for pin conflicts on heater pins.

SaschaKP, 2020-07-03:
recompiled with that define uncommented and the bug is gone, this is the output of the command, and I can't really find any conflicts, will do some more tests later.

SENT: M503
READ: echo:  G21    ; Units in mm (mm)
G21    ; Units in mm (mm)
READ: echo:  M149 C ; Units in Celsius
M149 C ; Units in Celsius
READ: 
READ: echo:; Filament settings: Disabled
; Filament settings: Disabled
READ: echo:  M200 T0 D1.75
M200 T0 D1.75
READ: echo:  M200 T1 D1.75
M200 T1 D1.75
READ: echo:  M200 S0
M200 S0
READ: echo:; Steps per unit:
; Steps per unit:
READ: echo: M92 X94.14 Y94.14 Z400.00 E140.00
M92 X94.14 Y94.14 Z400.00 E140.00
READ: echo:; Maximum feedrates (units/s):
; Maximum feedrates (units/s):
READ: echo:  M203 X250.00 Y250.00 Z20.00 E100.00
M203 X250.00 Y250.00 Z20.00 E100.00
READ: echo:; Maximum Acceleration (units/s2):
; Maximum Acceleration (units/s2):
READ: echo:  M201 X1000.00 Y1000.00 Z100.00 E5000.00
M201 X1000.00 Y1000.00 Z100.00 E5000.00
READ: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
READ: echo:  M204 P250.00 R5000.00 T500.00
M204 P250.00 R5000.00 T500.00
READ: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
READ: echo:  M205 B20000.00 S0.00 T0.00 J0.64
M205 B20000.00 S0.00 T0.00 J0.64
READ: echo:; Home offset:
; Home offset:
READ: echo:  M206 X10.00 Y10.00 Z-20.00
M206 X10.00 Y10.00 Z-20.00
READ: echo:; Hotend offsets:
; Hotend offsets:
READ: echo:  M218 T1 X34.00 Y0.00 Z0.000
M218 T1 X34.00 Y0.00 Z0.000
READ: echo:; Auto Bed Leveling:
; Auto Bed Leveling:
READ: echo:  M420 S1 Z0.00
M420 S1 Z0.00
READ: echo:  G29 W I0 J0 Z-0.27750
G29 W I0 J0 Z-0.27750
READ: echo:  G29 W I1 J0 Z-0.03000
G29 W I1 J0 Z-0.03000
READ: echo:  G29 W I2 J0 Z-0.09750
G29 W I2 J0 Z-0.09750
READ: echo:  G29 W I0 J1 Z-0.26750
G29 W I0 J1 Z-0.26750
READ: echo:  G29 W I1 J1 Z-0.02250
G29 W I1 J1 Z-0.02250
READ: echo:  G29 W I2 J1 Z-0.10750
G29 W I2 J1 Z-0.10750
READ: echo:  G29 W I0 J2 Z-0.25250
G29 W I0 J2 Z-0.25250
READ: echo:  G29 W I1 J2 Z0.03000
G29 W I1 J2 Z0.03000
READ: echo:  G29 W I2 J2 Z-0.21000
G29 W I2 J2 Z-0.21000
READ: echo:; Material heatup parameters:
; Material heatup parameters:
READ: echo:  M145 S0 H190 B60 F0
M145 S0 H190 B60 F0
READ: echo:  M145 S1 H240 B110 F0
M145 S1 H240 B110 F0
READ: echo:; PID settings:
; PID settings:
READ: echo:  M301 P20.03 I1.22 D82.05
M301 P20.03 I1.22 D82.05
READ: echo:  M301 P20.03 I1.22 D82.05
M301 P20.03 I1.22 D82.05
READ: echo:; Z-Probe Offset (mm):
; Z-Probe Offset (mm):
READ: echo:  M851 X14.00 Y-47.10 Z-2.48
M851 X14.00 Y-47.10 Z-2.48
READ: echo:; Filament load/unload lengths:
; Filament load/unload lengths:
READ: echo:  M603 T0 L0.00 U80.00
M603 T0 L0.00 U80.00
READ: echo:  M603 T1 L0.00 U80.00
M603 T1 L0.00 U80.00
READ: echo:; Tool-changing:
; Tool-changing:
READ: echo: Z2.00
Z2.00
READ: ok

sjasonsmith, 2020-07-05:
@SaschaKP, do you actually re-install the old firmware to verify it still works after the new firmware failed? I just want to rule out hardware damage, such as a burned-out mosfet.

grauerfuchs, 2020-07-05:
You said you're using the Arduino IDE as a compiler? Did you modify the pin definitions in the core libraries or switch to the MightyCore library for premade support? 

By default, Arduino doesn't properly address all of the pins used by the MightyBoard. I had the same issue early on when trying to compile Marlin builds for the MightyBoard, and that affected both PIO and Arduino at the time. 

Since then, the code has been updated (for PIO) to make it to use a board variant (MightyCore) that supports the needed pins. If the original firmware still works, try switching your compile environment to Platform IO instead of the Arduino compiler. If you need to run this entirely from command line, you'll need to update the pins definition.

SaschaKP, 2020-07-05:
that is what solved my issue, probably, since also older firmware worked, newer worked too, what probably solved the issue for the recompile for the new firmware is that the older one requested me with an error on compile time to modify that arduino pin definition file. so after that everything worked out.

I suppose that by giving the error in compile time, about number of pins, should fix future issues too, since older firmware do that and new one don't.

sjasonsmith, 2020-07-05:
I suggest leaving it open so that a compiler error can be added to the Mightyboard pins file. If there is an easy way to check the hardware it might help somebody else down the line.

SaschaKP, 2020-07-05:
> You said you're using the Arduino IDE as a compiler? Did you modify the pin definitions in the core libraries or switch to the MightyCore library for premade support?
>
> By default, Arduino doesn't [...]

I have to check that MightyCore, because Replicator 1 clones and Replicator 2 clones use different pins, maybe it's because of that, and by reassigning pins manually in arduino IDE that won't be called.

grauerfuchs, 2020-07-06:
So far, at least in my experience, the configuration now provided for use with PlatformIO works on the Mightyboard rev D that uses the ATMega2560. I have personally tested this and it works with a stock FlashForge Creator Pro and on the clone board removed from a CTC Replicator Dual. I have not tested it on the ATMega1280-based Mightyboard.

The library name "MightyCore" may not be correct; With a bit of additional digging done today, it looks like the library listed on github does not have support for the 1280 or 2560. Odd. Some of the documentation I have previously used may have incorrectly named the source of the board variant. Regardless of whether or not that is the official library in use, the PIO configuration present in the 2.0.x-bugfix branch does work, at the time of my last compile, without the requirement of altering the pins file. Unfortunately, the PIO configuration does not apply to compiles done with the Arduino IDE.

SaschaKP, 2020-07-06:
Mightyboard with ATMEGA1280 or 2560 only vary on the programmable memory size (double size on 2560), all the rest is exactly the same on both controllers, like SRAM and other things.

grauerfuchs, 2020-07-06:
That's what I had understood of the configuration, but I haven't reviewed the MakerBot or Sailfish code to verify that there have not been any pin changes.

SaschaKP, 2020-07-07:
anyway, the fact is that arduino IDE only maps the first 70 pins of the arduino board, while mightyboard do use 80 pins in the map, the latest 10 are unrecognized by the compiler itself.

The non-bugfix version will give an error on compile time, requesting a modification on those core files, while the latter will not request, making those pins non functional, since they are not configured and used in the correct way in the various defines. that's probably what is happening here.

boelle, 2020-07-08:
i would recommend to go the vscode+platform.io route, once you have tried you will not go back to arduino ide

ellensp, 2020-07-08:
when you compiled, what did you select in arduino ide as the board type?

According to https://cdn.thingiverse.com/assets/6d/94/65/9b/90/MakerBot_MightyBoard_REVE_Schematic.pdf 
the hotend pins are H3 and B5
The provided m43 output lists
READ: PIN:   6   Port: H3        HEATER_0_PIN                           protected 
READ: PIN:  11   Port: B5        HEATER_1_PIN                           protected 
Shows these to be correct.

These are standard arduino pins.  nothing special about them.

SaschaKP, 2020-07-08:
the first page you see is exactly what I had for pins config, and never changed the mightyboardreve.h, the arduino type is a atmega2560, as selected in the arduino IDE, before and after.

It can't be that the problem has gone itself as if nothing did happen, that's not truly possible.

I will try to recreate the situation, and if successfull I'll post it here, negative or positive case.

grauerfuchs, 2020-07-08:
@ellensp - It's been a long time since I've gone over the details for where and how the underlying issue occurs, but if I recall correctly, it's due to the unpredictable nature of reading data past the end of the pins array (out-of-bounds). In my case and with the code I had at the time, it resulted in the outward appearance of E1 (L) not heating up. The issue also isn't universal; It only occurs in functions that use certain calls in Arduino-based libraries. I'd need to go back through and find where the issue was initially documented to give details beyond this, but it's a long-standing, known issue with the Arduino framework/compiler/IDE with respect to the MightyBoard. Thankfully, the issue has been resolved when compiling the bugfix branch using VSCode and PIO due to its support for the MegaCore library and the extended pins definitions in the board config.

ellensp, 2020-07-08:
I wonder if its that power cut off circuit as it can cut off voltage to hotend.
try a m42 p16 s1  as that should reset them on.
then try heating up.

SaschaKP, 2020-07-27:
It currently works, and I'm currently using the machine, tested with marlin 2.0.x-bugfix, recently updated.

What I've found out is that arduino IDE don't declare all the pins used by a mightyboard REV-E (so this requires a pin readdressing), so at this point the current bugfix-2.0.x works correctly with that board on an avr-atmega2560, this can be considered solved at the moment.

this is what I found on an old version of marlin source, and applied all the changes, and it currently works, it could be as @grauerfuchs  stated, it only happens when certain functions are called, declaring those missing pins in the arduino IDE will make it work everything, but currently, marlin 2.0.x-bugfix don't have that disclaimer/error:
`
```c
// Check if all pins are defined in mega/pins_arduino.h
#include <Arduino.h>
static_assert(NUM_DIGITAL_PINS > MAX_PIN, "add missing pins to [arduino dir]/hardware/arduino/avr/variants/mega/pins_arduino.h based on fastio.h"
                                          "to digital_pin_to_port_PGM, digital_pin_to_bit_mask_PGM, digital_pin_to_timer_PGM, NUM_DIGITAL_PINS, see below");

/* in [arduino dir]/hardware/arduino/avr/variants/mega/pins_arduino.h
change:
#define NUM_DIGITAL_PINS            70
to:
#define NUM_DIGITAL_PINS            80

to digital_pin_to_port_PGM add at the end:
const uint8_t PROGMEM digital_pin_to_port_PGM[] = {
....
        PG      , // PG 4 ** 70 **
        PG      , // PG 3 ** 71 **
        PJ      , // PJ 2 ** 72 **
        PJ      , // PJ 3 ** 73 **
        PJ      , // PJ 7 ** 74 **
        PJ      , // PJ 4 ** 75 **
        PJ      , // PJ 5 ** 76 **
        PJ      , // PJ 6 ** 77 **
        PE      , // PE 2 ** 78 **
        PE      , // PE 6 ** 79 **
};

to digital_pin_to_bit_mask_PGM  add at the end:
const uint8_t PROGMEM digital_pin_to_bit_mask_PGM[] = {
....
        _BV( 4 )        , // PG 4 ** 70 **
        _BV( 3 )        , // PG 3 ** 71 **
        _BV( 2 )        , // PJ 2 ** 72 **
        _BV( 3 )        , // PJ 3 ** 73 **
        _BV( 7 )        , // PJ 7 ** 74 **
        _BV( 4 )        , // PJ 4 ** 75 **
        _BV( 5 )        , // PJ 5 ** 76 **
        _BV( 6 )        , // PJ 6 ** 77 **
        _BV( 2 )        , // PE 2 ** 78 **
        _BV( 6 )        , // PE 6 ** 79 **
};

to digital_pin_to_timer_PGM add at the end:
const uint8_t PROGMEM digital_pin_to_timer_PGM[] = {
....
        NOT_ON_TIMER    , // PG 4 ** 70 **
        NOT_ON_TIMER    , // PG 3 ** 71 **
        NOT_ON_TIMER    , // PJ 2 ** 72 **
        NOT_ON_TIMER    , // PJ 3 ** 73 **
        NOT_ON_TIMER    , // PJ 7 ** 74 **
        NOT_ON_TIMER    , // PJ 4 ** 75 **
        NOT_ON_TIMER    , // PJ 5 ** 76 **
        NOT_ON_TIMER    , // PJ 6 ** 77 **
        NOT_ON_TIMER    , // PE 2 ** 78 **
        NOT_ON_TIMER    , // PE 6 ** 79 **
};
*/
````

gilesdavison, 2020-08-04:
Thought I should chime in on this as my issue is identical. I have the ATMega1280 with the same heater pin out H3 and B5 all wired up correctly and no hardware issues (previously ran @SaschaKP 's modified Sailfish version with everything working). Currently running the latest bugfix and compiling with VSCode instead of IDE. My T0 (right side extruder) is also not heating up. What's even more odd is that when I use the filament loading regardless of choosing E1 or E2, the right extruder (T0) always heats up and the left... nothing.

SaschaKP, 2020-08-04:
@gilesdavison 
use this commit: b0c6cfb0511b8d71fec26ab91cd945f3d623b3c0
after this we have a problem on multiple toolhead

gilesdavison, 2020-08-05:
@SaschaKP Which specific commit are you using? I've tried the previous commit to [b0c6cfb](https://github.com/MarlinFirmware/Marlin/commit/b0c6cfb0511b8d71fec26ab91cd945f3d623b3c0) and I'm still having problems.. I then tried the 2.0.5 release and it's working but as you mentioned there are probe issues with that release.

boelle, 2020-08-05:
> Arduino 1.8.13 as avr compiler,

try vscode+plaform.io

boelle, 2020-08-05:
@SaschaKP did you try vscode?

SaschaKP, 2020-08-05:
@boelle No, I still use arduino IDE with modified pins_arduino.h, but in arduino IDE the progmem won't be sufficient for an ATMEGA1280, 128KB of progmem are mostly used by the auto-leveling and all the strings, I'll try later with the gcc cross toolchain

boelle, 2020-08-06:
thinking of it this question came up in my head: why not vscode+platform.io?

ellensp, 2020-08-14:
vscode+platformio has same missing pins issue
Everything is based on Arduino mega pin definitions and they do not use pin > 69
for platformio you can find the file in .platformio/packages/framework-arduino-avr/variants/mega/pins_arduino.h

boelle, 2020-09-11:
@SaschaKP tried what @ellensp suggested?

> But, you are meant to use env: MightyBoard1280 or env: MightyBoard2560 env, these use the ATmega2560 megacore framework. And has the extra pins.

closed 2020-09-23 by github-actions[bot]