MarlinFirmware/Marlin #18523: [BUG] Build Error After the latest 15 commits

KL-XL, 2020-07-03:
### Bug Description

After merging the last 15 commits from Bugfix-2.0x  this morning the firmware build errors out if Creality/Ender-3/BigTreeTech SKR Mini E3 2.0 config files are used.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/4869141/Config.Files.zip)

### Steps to Reproduce

Step 1: download the latest Marlin 2.0 Bugfix source code
Step 2: copy the config files from Creality/Ender-3/BigTreeTech SKR Mini E3 2.0 to Marlin folder
Step 3: change the default_envs value in platformio.ini to STM32F103RC_btt_512K
Step 4: press Build

**Expected behavior:** [What you expect to happen]

Should show build success 

**Actual behavior:** [What actually happens]

[image]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-07-03:
See PR https://github.com/MarlinFirmware/Marlin/pull/18519

KL-XL, 2020-07-03:
Thanks. this does the trick.

closed 2020-07-03 by KL-XL