MarlinFirmware/Marlin #18524: [BUG] 3 Point Bed Leveling should rize Z before moving to start position

bcdiaconu, 2020-07-03:
### Bug Description

At the moment, 3 point UBL mesh tuning would move the hot-end from whereever it is to the first point's {X,Y} position. Than rises the Z and starts measuring each point which sometimes creates collision of the nozzle with the bed.

### My Configurations

https://github.com/bcdiaconu/Marlin/blob/2.0.x/Marlin/Configuration.h
https://github.com/bcdiaconu/Marlin/blob/2.0.x/Marlin/Configuration_adv.h

### Steps to Reproduce

`G29 J` or menu -> Motion -> UBL -> UBL Tools -> Mesh Leveling -> 3 Point Leveling

**Expected behavior:** 

When sendigh the command either through printer menu or by G-Code, first it should rize the head to Z position and than move to {X, Y} position. This will prevent a situation where when the hardware min-position switch on G28 command will take nozzle lower to be lower than bed and therefore to create a collision, scrath or damage when G29 command is issued (before fine adjustment of UBL).

**Actual behavior:** 

At the moment, 3 Bed UBL mesh tuning would move the hot-end from whereever it is to the first point's {X,Y} position. Than rises the Z and starts measuring each point

ellensp, 2020-07-03:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

bcdiaconu, 2020-07-06:
@ellensp Thanks. I will try, but my config is not compatible with latest commit yet and my printer head is disassembled. When I'll put it together I'll let your know the results.

qwewer0, 2020-07-06:
G29 J does this for me:

1. Homes all axis
2. Moves Z axis to `Z_CLEARANCE_DEPLOY_PROBE`
3. Probes first point
4. Moves Z axis to `Z_CLEARANCE_BETWEEN_PROBES`
5. Probes second point
6. Moves Z axis to `Z_CLEARANCE_BETWEEN_PROBES`
7. Probes third point

So, the nozzle is always above the bed.

bcdiaconu, 2020-07-10:
If I send `G28` followd by `G29 J` or directly `G29 J`, it will still go to {X,Y} first and than rize the hotend.

Configs for bugfix_2.x branch:
https://github.com/bcdiaconu/Marlin/blob/bugfix-2.0.x/Marlin/Configuration.h
https://github.com/bcdiaconu/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h
https://github.com/bcdiaconu/Marlin/blob/bugfix-2.0.x/Marlin/ZonestarP802Qs.h

bcdiaconu, 2020-07-10:
One more observation after changin the `Configuration.h` (please see my latest commit for this file):

If I have enabled and set `Z_AFTER_PROBING` it will have totally different behaviour after starting the printer and sendign `G29 J`:
1. Homes all axis
2. Moves Z axis to `Z_CLEARANCE_BETWEEN_PROBES`
3. Move to first point
4. Moves Z axis to `Z_CLEARANCE_DEPLOY_PROBE`
5. Lower Z to Probes first point
6. Moves Z axis to `Z_CLEARANCE_BETWEEN_PROBES`
7. Probes second point
8. Moves Z axis to `Z_CLEARANCE_BETWEEN_PROBES`
9. Probes third point
10. Moves Z axis to `Z_CLEARANCE_BETWEEN_PROBES`

Please see steps 
- 2 -> 3 -> 4
- step 10 rizes to `Z_CLEARANCE_BETWEEN_PROBES` instead of `Z_AFTER_PROBING`

bcdiaconu, 2020-07-10:
Last obervation. Please see all my previous 3 messages:

With `Z_AFTER_PROBING` defined, if I send `G0 X0 Y0` followed by `G0 Z0` and `G29 J` the behaviour is the same as initial problem. It will not rise Z first.

closed 2020-10-20 by github-actions[bot]