MarlinFirmware/Marlin #18525: [BUG] Thermal runaway detected while using #999 & #998 virtual temp sensors

Patag, 2020-07-03:
### Bug Description

Thermal runaway is detected at startup and stops Marlin. 
### My Configurations
BTT SKR Pro v1.1 on test bench
[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/4869890/Conf.zip)

### Steps to Reproduce
1. Download Marlin bugfix 2.x after 30/06 (don't know exactly what commit)
2. define TEMP_SENSOR_0 999 and TEMP_SENSOR_BED 998
3. Compile and install
2. Restart

**Expected behavior:** Normal Marlin startup

**Actual behavior:** Thermal runaway detection "MAX TEMP on E1" or something like that.

#### Additional Information
This happend after 30/06 commit

ellensp, 2020-07-03:
If you give any heating gcodes you can still trigger a Thermal runaway.

Patag, 2020-07-03:
No gcodes are sent at all

ellensp, 2020-07-03:
Confirmed on a standard ramps current bugfix.  Only config changes was to enable a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and set TEMP_SENSOR_0 999 and TEMP_SENSOR_BED 998,

ellensp, 2020-07-04:
Seem to have been introduced int this commit. https://github.com/MarlinFirmware/Marlin/commit/c43bbcce152b4d3b4697e1e188d3bcada76a24a7

ellensp, 2020-07-04:
See the PR for how to get around this issue.

Patag, 2020-07-04:
Works great. Thank you. Do you want me to close this issue or do we wait for official merge ?

sjasonsmith, 2020-07-04:
@Patag I suggest you wait for the PR to merge.

sjasonsmith, 2020-07-04:
Ideally you would re-test after the merge to be sure it is fixed. Sometimes PRs change during the PR process and are merged before bugs are found.

Patag, 2020-07-04:
OK

setar, 2020-07-04:
https://github.com/MarlinFirmware/Marlin/pull/18533/commits/8a594fc7a0721181351c15dc0c249022c7999d60 - Not fixed all problem:
I have the same bug, MSG_ERR_MAXTEMP on E1
But i not use E1 and not use #999 and #998 sensors .

#define TEMP_SENSOR_0 1047
#define TEMP_SENSOR_1 0
--skip--
#define TEMP_SENSOR_BED 1
#define TEMP_SENSOR_PROBE 0
#define TEMP_SENSOR_CHAMBER 13

#define TEMP_BED_PIN                    P0_23_A0  // A0 (TH1)
#define TEMP_0_PIN                      P0_24_A1  // A1 (TH2)
#define TEMP_CHAMBER_PIN                P0_25_A2  // A2 (TH3)

sjasonsmith, 2020-07-05:
@setar, please do not add your unrelated issue to this one. This issue is specific to simulator sensors 998 and 999.

@Patag, this has now been merged. There were some changes made prior to merging, so you may want to re-test it before closing the issue.

ellensp, 2020-07-06:
I have just retested with merged changes. They work fine.
So i'm closing this.
@setar  If your unrelated issues still persists then please open a new issue.

closed 2020-07-06 by ellensp