MarlinFirmware/Marlin #18526: [BUG] BLtouch Occasionally and randomly triggers incorrectly during probing.

The3DE, 2020-07-03:
### Bug Description

I am running the SKR mini E3 v2.0 with a BLtouch v3.1 and I am having problems with probing. Most of the time, probing is successful, but occasionally the probe fails to trigger correctly. The pin deploys, but when it contacts the bed, it doesn't trigger and retract, so the gantry continues to move down. Once it goes for about a millimeter, the probe flashes red a couple times and triggers, but it triggers over 1mm too low. This happens randomly and not very often. I am using UBL and when probing a 10x10 mesh about 1 point out of the 100 is incorrect each time. It also happened once during a G28 autohome when I was watching. I have verified that the wiring is correct, have tested 2 different BLtouch probes and cables, and checked that the 5v supply on the board is working, as well as changing various parts of the firmware, but the issue persists. If the probing is successful, everything else works, but I cant seem to stop it from not triggering correctly that ~1% of the time.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4870172/Configurations.zip)

### Steps to Reproduce

1. Wire and set up BLtouch v3.1 with the SKR mini E3 v2.0
2. Test probing (I did this with the 10 point mesh in UBL)

**Expected behavior:** [What you expect to happen]
All points trigger correctly and return the correct values.
**Actual behavior:** [What actually happens]
Sometimes 1-3 points of the 100 trigger incorrectly and the returned value is over 1mm lower than expected.
#### Additional Information
Tried 2 different probes, cables, checked all the connections, and commented/uncommented various features and combinations and firmware with no effect. Error is random and not frequent (~1%) but doesn't go away. An incorrect probe during autohome will mess up the z offset.
[image]

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

GMagician, 2020-07-03:
I suggest you to enable `ENDSTOP_INTERRUPTS_FEATURE` and give a try.
OT: how did you get the graph you posted?

The3DE, 2020-07-03:
I’ll try to enable interrupts but I don’t know if it would change anything since the BLTouch itself doesn’t trigger at those points, and it’s not that the board doesn’t detect the trigger, at least that’s what I think.

I got the graph from the bed visualizer plugin in octoprint.

GMagician, 2020-07-03:
If bl-touch doesn't trigger I suspect Marlin can do nothing to solve it

The3DE, 2020-07-03:
It only occasionally happens and I’ve gone through 2 different BLTouches as I though the sensor was defective, both both show the same behavior and others have reported this problem as well. I think that it is a problem with marlin sending commands to the BLTouch.

GMagician, 2020-07-03:
mmmm now sure but when Marlin stow bl-touch doesn't trigger error (red led blink) it seems like the trigger is missed and interrupt may help

qwewer0, 2020-07-03:
I have the same issue, with BLTouch v3.1, SKR Mini E3 v1.2.

Standard deviation 0.0015+-0.0005
With `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` enabled, 
[BLTouch ALARM or TRIGGER after STOW recovering-M48_S1_V4.txt](https://github.com/MarlinFirmware/Marlin/files/4870581/BLTouch.ALARM.or.TRIGGER.after.STOW.recovering-M48_S1_V4.txt)

Will test `ENDSTOP_INTERRUPTS_FEATURE` and give feedback.

GMagician, 2020-07-03:
ok I see it:
READ: BLTouch STOW requested
READ: BLTouch Command :90
READ: BLTouch ALARM or TRIGGER after STOW, recovering
READ: BLTouch Command :160
READ: BLTouch Command :90
READ: bltouch.stow_proc() end

qwewer0, 2020-07-03:
With `ENDSTOP_INTERRUPTS_FEATURE` enabled, I run `M48 P50` six times without any bltouch alarm. (4 with `BLTOUCH_HS_MODE` and 2 without.)
Can't say that it fixed the issue until I made a lot more probing with different speeds and settings.

qwewer0, 2020-07-03:
Got a bltouch error, but it is not the same as before:
```
READ: BLTouch STOW requested
READ: BLTouch Command :90
READ: bltouch.stow_proc() end
READ: >>> do_blocking_move_to X158.8600 Y121.0700 Z7.4850
READ: <<< do_blocking_move_to
READ: //action:notification Probing Failed
READ: Error:Probing Failed
READ: <<< Probe::probe_at_point
READ:   current_position= X158.8600 Y121.0700 Z7.4850 : Probe::set_deployed
READ: deploy: 0
```

GMagician, 2020-07-03:
This error seems related to probe not deployed when required. Can you post something before?

qwewer0, 2020-07-03:
@GMagician Thanks for the help.
Here is the whole M48: 
[M48 V4 P50.txt](https://github.com/MarlinFirmware/Marlin/files/4870728/M48.V4.P50.txt)

GMagician, 2020-07-03:
here it is:
READ: SLOW Probe fail! - No trigger.

qwewer0, 2020-07-03:
Thanks, did not saw that.
But I don't really know what any of the fails could mean.

GMagician, 2020-07-03:
not sure but maybe increase a little bit Z_PROBE_LOW_POINT. It can also be that ibl-touch doesn't trigger for real...

qwewer0, 2020-07-03:
By 'increase' you mean decrease, right?

The3DE, 2020-07-03:
I’ve just started testing with endstop interrupts enabled and so far it’s successfully probed a 10 point grid and a standard M48 with 10 points gives a SD of ~.0011. I’ll continue testing to see if any issues come up. I enabled 5V mode and set z_min_probe_endstop_inverting to false. I also just successfully completed a M48 P50. High speed mode is off.
When the printer powers on the BLTouch does it self test but then deploys, flashes red a few times quickly, and then retracts and goes back to normal, is this behavior supposed to happen?

qwewer0, 2020-07-03:
The quick flashes at the start is the indication of the 5V mode.

qwewer0, 2020-07-03:
Before I had 1-2 failed probing out of 100, but now I had 0 out of +1000 (M48), and was able to probe 7*7 UBL mesh two times without any issue.
So far so good.

@GMagician Could you please explain why could have ENDSTOP_INTERRUPTS_FEATURE helped, and what it does?

> Enable this feature if all enabled endstop pins are interrupt-capable.

And how can I check if I would be able to use it on a different board?

The3DE, 2020-07-03:
I am guessing that the board has missed the trigger pulse from the BLTouch, which caused the previous problems. Interrupts would fix this as it would be able to sense the change in state of the pin reliably. I also am having very good success after enabling this, so far no fails on a few 10x10 meshes in UBL and a few M48s.

ManuelMcLure, 2020-07-03:
Another option is to enable ``BLTOUCH_FORCE_SW_MODE`` which makes the BLTouch always show triggered when retracted instead of just generating a short pulse. This makes it act more like a conventional micro switch endstop.

qwewer0, 2020-07-03:
> Another option is to enable BLTOUCH_FORCE_SW_MODE which makes the BLTouch always show triggered when retracted instead of just generating a short pulse. This makes it act more like a conventional micro switch endstop.

@ManuelMcLure Will try that too, thanks.

GMagician, 2020-07-03:
@qwewer0 interrupt is a fast way to detect input. Once input pin changes it's state cpu is redirected to check it. Standard mode checks input (endstop) on a 1KHz base frequency (into temperature code) so you may miss a signal that changed very quickly (bltouch trigger)

and yes increase the number but since it's negative decrease is a best description. Sorry for that

The3DE, 2020-07-04:
I 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4872245/Configuration.zip)
I don't know why but the triggering problem came back. I have attached my current configs. I have interrupts enabled and it seemed to work, but then I switched to a different sensor and it failed to trigger again. I then switched back to the sensor that had worked but it didn't fix the problem. I have 5v mode enabled, but the probe no longer does the deploy, flash, and stow sequence at the start for some reason.

I also tried enabling force SW mode, and I briefly tested it without failure, but it caused my standard deviation to be > .01, around >03 from what I remember, so I switched it back off.

GMagician, 2020-07-04:
5V mode... is your board a 5V tolerant input?

GMagician, 2020-07-04:
> Should I try decreasing z_probe_low_point or is that for another issue?

May be increased when probe stow & deploy correctly but Z moves down and probe doesn't reach the "trigger position" you have to see a failure like the one posted above. It seems not your situation.
Now I don't know what is different between SW and 5V mode on bltouch, I have a old V2.0 (clone), I need to read documentation about it.
But all these seems a configuration issues, not related to Marlin. I think it's better to move to discord (but I'm not there often)

The3DE, 2020-07-04:
The problem is that I'm not really sure what the issue really is, as I couldn't find much about it when doing a search. My board is 5V compatible. 
I am pretty sure that SW mode just causes the trigger pulse to be longer and act like a conventional switch, and that 5V mode doesn't effect it.
I don't know whether the probe itself isn't triggering correctly or the mainboard isn't detecting the trigger pulse, and at this point I'm not even sure if its a hardware or software issue, even though I tested 2 separate BLTouches. I wish I had a video of it failing but it is random and infrequent so I can't predict when it'll fail and I don't want to record a whole M48 P50 and hope that I catch it.
I can try asking on discord to see if anyone else has any idea.

kmobs, 2020-07-04:
I had this issue consistently on the 1.2 version of this board. The stock configs puts the stepper current very high in the config. I dropped the x and y to 580 and the z to 600. I also recommend making sure your belts aren't too tight. I think it's power related with the BLtouch pin. 

After changing these things I get near 0 failures.

GMagician, 2020-07-04:
@kmobs yours can be a solution. PWM and high current generates electric noise.
High drain may lower 5V if PSU is not "big" enough or board has bad traces and bl touch may have issues...

cjsoong, 2020-07-04:
@GMagician 
You can add a 10~22uf capacitor to the 5v & GND of BLtouch, which can stabilize the BLtouch power supply..

> @kmobs yours can be a solution. PWM and high current generates electric noise.
> High drain may lower 5V if PSU is not "big" enough or board has bad traces and bl touch may have issues...

kmobs, 2020-07-04:
@cjsoong that's a good idea. Not sure if this is truly the reason, but seems to have mostly solved my issues. Hopefully not just a coincidence

The3DE, 2020-07-04:
My stepper current is currently set to 580 on all 3 axis. I can try putting a capacitor on the power pins for the BLTouch.
Are you running the BLTouch in force sw mode?

The3DE, 2020-07-04:
I put a 10uF electrolytic capacitors across the 5V lines and it didn’t fix the problem. I’ll leave it on there since it probably doesn’t hurt but I don’t think it’s an issue with the power supply

The3DE, 2020-07-05:
I uncommented force SW mode and so far no fails. The SD is greater than before (.012 vs .002 before), and the range has increased to ~.07. I think this means that where it would previously fail, since it is in SW mode, the board can detect the trigger, but it detects it a bit late, so the trigger point is about .07mm lower than it should be. This is improved from it completely missing the trigger, but I'm wondering if there is any way to decrease this inaccuracy. I have already slowed probing speeds down and it seems to be moving pretty slow, any slower and probing would take way too long. The only way I can think of to work around this is to enable multiple and extra probing in order to invalidate the inconsistent probes, along with slowing down the homing validation speed by a lot.

The3DE, 2020-07-05:
It seems that the board is slow to respond to the pin’s change in state sometimes, even with interrupts enabled, any idea what’s causing this?

kmobs, 2020-07-05:
@The3DE no I'm not using SW mode. I did force 5V mode though. Look at my repo for my current config

GMagician, 2020-07-05:
I read that with big condenser on input may give problems to bl-touch. Sure is not your case? on antclabs site there are some fixes to apply in such cases

sjasonsmith, 2020-07-05:
It seems like we have two possible solutions for this:

1. Enable `ENDSTOP_INTERRUPTS_FEATURE`, although it might not be an option for all boards.
2. `BLTOUCH_FORCE_SW_MODE`

Is anybody still experiencing issues with one of these enabled, or can this be closed as a configuration issue?

The3DE, 2020-07-05:
@GMagician a capacitor on the 5V power lines doesn’t effect it, that refers to the capacitor on the sensor input. The dedicated probe pin for the BLTouch doesn’t have this capacitor, so I might try to switch the zmin endstop pin and the probe pin in the pin configurations and plug the BLTouch into the probe pin to see if it helps.

I am not getting any failed probes, but I have discovered that the inaccuracies I am getting is from it occasionally probing about .07mm lower than expected. This is way better than the ~1.3mm from before, As it doesn’t cause the nozzle to crash, but .07mm is still somewhat troubling as that can be half a layer height. I will try to test some things to see if this can be reduced, but enabling multiple and extra probing would be able to work around it. I believe that this is the result of the board occasionally being slow to detect and respond to the change in the endstop pin, even with interrupts enabled. Other than those times it probes low, the BLTouch is very accurate and repeatable.

I also still haven’t ruled out that this is an issue with my board and not the firmware or configs, so I got a capacitive probe and I will test it to see if it has the same problems as the BLTouch, in which case my board might just be defective, or this issue applies to z probes in general and is not exclusive to the BLTouch.

GMagician, 2020-07-05:
@The3DE the capacitor on 5V supply is to keet 5V stable when someone drain current. The one I was refering to was on input but since you don't have it, it's better. Don't move to endstop. 
With interrupt is really hard to think board is "slow" to detect (only reason is an interrupt with higher priority).

The3DE, 2020-07-05:
I looked at the schematic and the SKR mini E3 v2.0 has a .1uF capacitor on the z endstop input. The probe input, however, doesn’t have this capacitor. That is why I thought of trying to redefine the probe input as the z endstop input in the pin configs and using that to see if anything changes.

As far as the interrupts, I am puzzled by that too, but I have seen others get better results with interrupts disabled, so I might try that to see if anything changes.

kursatu, 2020-07-06:
My 2cents. I also had such problems with Skr 1.3, 1.4, & turbo boards, bl touch dropping pin, not dropping during probing etc. Tried bunch of stuff , capacitors, supercapacitors, SW_MODE, looked at the defect related rto LPC boards, platform code etc.
It turned out to be EMI (elctromagnetic interference), well at least portions of it. 
The problem was quickly reproducible.
Changed the heater cable to be  twisted pair I've made from speaker wires , it helped a lot. 
I am still waiting twisted pair cables to replace all the other wires. It sill happens but a lot less.

The3DE, 2020-07-06:
@kursatu Did this only happen with heaters on? When I tested the BLTouch most of the time it was with the bed and hotend off.

ManuelMcLure, 2020-07-06:
One thing to check is if the servo port that the BLTouch is connected to outputs 5V or 3.3V signals. If it outputs 3.3V this could be a bit too low for the BLTouch. That's one reason I added a 3.3V->5V level shifter when I connected my BLTouch to my Re-ARM/RAMPS combination.
If nothing else, using 5V signals should make them more resistant to interference.

The3DE, 2020-07-06:
I'm pretty sure that the SKR is a 3.3v board as it is based off a STM32, but it can work with a BLTouch in force 5v mode, so I'm not sure. I looked at the schematic and pinouts and the BLTouch connections don't go to 5v tolerant pins, but it works in 5v mode, and is even default in the example Marlin has for the board. However, I am pretty sure that the board still outputs 3.3v on all pins, so I might try to add a level shifter to see if that helps.

kursatu, 2020-07-06:
@The3DE  Yes, it did happen with the heaters. I was able to conclusively isolate it as a contributor.
Your case might differ, apparently.

ManuelMcLure, 2020-07-06:
In this case it’s not the output signal from the BLTouch that’s the problem but rather the input signal that controls it. The BLTouch is powered from 5V so it’s expecting 5V TTL signal levels and 3.3V is just on the edge of 5V logic high so it may work sporadically.

qwewer0, 2020-07-06:
With `ENDSTOP_INTERRUPTS_FEATURE` I get error less often, and maybe a different kind.
Here is one with `BLTOUCH_HS_MODE`:

```
READ: SLOW Probe fail! - No trigger.
READ:   current_position= X158.8600 Y121.0700 Z-0.5150 : <<< run_z_probe
READ:   current_position= X158.8600 Y121.0700 Z-0.5150 : Probe::set_deployed
READ: deploy: 0
READ: Probe::move_z(6.0000)
READ: >>> do_blocking_move_to X158.8600 Y121.0700 Z7.4850
READ: <<< do_blocking_move_to
READ: BLTouch STOW requested
READ: BLTouch Command :90
READ: bltouch.stow_proc() end
READ: >>> do_blocking_move_to X158.8600 Y121.0700 Z7.4850
READ: <<< do_blocking_move_to
READ: //action:notification Probing Failed
READ: Error:Probing Failed
```

With `BLTOUCH_FORCE_SW_MODE` I didn't get errors yet, but that doesn't mean anything, as at first that was my experience with `ENDSTOP_INTERRUPTS_FEATURE` too, so we will see in the long run.
But what bugs me is that the standard deviation goes from ~0.002 to ~0.02 with `BLTOUCH_FORCE_SW_MODE`.

The3DE, 2020-07-06:
@qwewer0 I have that same problem with standard deviation in SW mode too and it is also bugging me. I haven't had any fails yet, but the SD has increased to around the same as yours. Looking at the actual data from the probes, it is mostly consistent, but occasionally there is an outlier that causes the SD to be so high. What is the range you are getting, along with the min and max? My range with SW mode on is about .07, with max being around 0 and min being around -.07. I looked at the points and sometimes they were around .07mm too low. I think this is because SW mode stops the probe and board from missing the trigger completely, but it is still late to detect the trigger, so the late reaction causes the point to be around .07mm too low. This means that SW mode makes the previous problem a lot better, but still doesn't solve it completely as it is still probing outliers.

GMagician, 2020-07-06:
As stated before I don't have an original bl-touch (but I will, an original 3.1). I read on web site that new 3.1 has a programmable trigger duration.
Since, AFAIK, sw mode works like a switch (then I think once it triggers it keeps the signal) and this mode has SD issues (I don't find a reason for that), other mode may be programmed to avoid trigger miss but I'm not sure Marlin support duration programming

The3DE, 2020-07-06:
@GMagician I don’t think that the standard deviation issues are because of SW mode, but actually still with the board sensing the BLTouch trigger. In my comment above, it says that the reason the SD is high is because there are outlier low points. Increasing the trigger duration would lead to the same issue in my opinion.

qwewer0, 2020-07-06:
> What is the range you are getting, along with the min and max?

```
READ: Bed X: 116.0000 Y: 115.0000 Z: -0.0025
READ: 1 of 9: z: -0.005 mean: -0.0050 sigma: 0.000000 min: -0.005 max: -0.005 range: 0.000
READ: 2 of 9: z: -0.072 mean: -0.0387 sigma: 0.033750 min: -0.072 max: -0.005 range: 0.067
READ: 3 of 9: z: -0.003 mean: -0.0267 sigma: 0.032425 min: -0.072 max: -0.003 range: 0.070
READ: 4 of 9: z: -0.003 mean: -0.0206 sigma: 0.029967 min: -0.072 max: -0.003 range: 0.070
READ: 5 of 9: z: -0.003 mean: -0.0170 sigma: 0.027767 min: -0.072 max: -0.003 range: 0.070
READ: 6 of 9: z: -0.005 mean: -0.0150 sigma: 0.025739 min: -0.072 max: -0.003 range: 0.070
READ: 7 of 9: z: -0.005 mean: -0.0136 sigma: 0.024085 min: -0.072 max: -0.003 range: 0.070
READ: 8 of 9: z: 0.000 mean: -0.0119 sigma: 0.022972 min: -0.072 max: 0.000 range: 0.072
READ: 9 of 9: z: -0.005 mean: -0.0111 sigma: 0.021766 min: -0.072 max: 0.000 range: 0.072
READ: Finished!
READ: Mean: -0.011111 Min: -0.072 Max: 0.000 Range: 0.072
READ: Standard Deviation: 0.021766
```

The3DE, 2020-07-06:
So you are getting the same results as me, where the BLTouch occasionally probes low by around .07mm. Anyone know how to fix this?

sjasonsmith, 2020-07-06:
Do you have `MONITOR_DRIVER_STATUS` enabled in `Configuration_adv.h`? If so, try disabling it.

The3DE, 2020-07-06:
@sjasonsmith I do have MONITOR_DRIVER_STATUS enabled, and I can try disabling it, but would doing so be a safety concern/ risk damaging the Trinamic drivers?

sjasonsmith, 2020-07-06:
It mainly won’t reduce current if they are getting too hot. I never enable it, because it is a lot of overhead. Looking at the code, disabling probably won’t help the standard deviation, but it should be easy to test.

qwewer0, 2020-07-06:
Tried it with MONITOR_DRIVER_STATUS disabled, and it gives kind of the same result.
```
READ: Bed X: 116.0000 Y: 115.0000 Z: -0.0050
READ: 1 of 9: z: -0.005 mean: -0.0050 sigma: 0.000000 min: -0.005 max: -0.005 range: 0.000
READ: 2 of 9: z: -0.003 mean: -0.0038 sigma: 0.001250 min: -0.005 max: -0.003 range: 0.002
READ: 3 of 9: z: -0.115 mean: -0.0408 sigma: 0.052454 min: -0.115 max: -0.003 range: 0.112
READ: 4 of 9: z: -0.005 mean: -0.0319 sigma: 0.048003 min: -0.115 max: -0.003 range: 0.112
READ: 5 of 9: z: -0.003 mean: -0.0260 sigma: 0.044514 min: -0.115 max: -0.003 range: 0.112
READ: 6 of 9: z: -0.005 mean: -0.0225 sigma: 0.041382 min: -0.115 max: -0.003 range: 0.112
READ: 7 of 9: z: -0.005 mean: -0.0200 sigma: 0.038799 min: -0.115 max: -0.003 range: 0.112
READ: 8 of 9: z: -0.005 mean: -0.0181 sigma: 0.036631 min: -0.115 max: -0.003 range: 0.112
READ: 9 of 9: z: -0.003 mean: -0.0164 sigma: 0.034883 min: -0.115 max: -0.003 range: 0.112
READ: Finished!
READ: Mean: -0.016389 Min: -0.115 Max: -0.003 Range: 0.112
READ: Standard Deviation: 0.034883
```
I don't really like the SD value, but the inconsistency is the worse.

GMagician, 2020-07-06:
> In my comment above, it says that the reason the SD is high is because there are outlier low points. Increasing the trigger duration would lead to the same issue in my opinion.

I'm think I'm lost...

The3DE, 2020-07-06:
@GMagician if you look at the data from qwewer0 you can see that most of the probes are accurate, but there is one probe that is too low. That’s is the reason the standard deviation is high. I don’t think that this is an issue with SW mode for the BLTouch, but actually with the board being late to respond to the probe signal, which would mean that reprogramming the pulse length wouldn’t fix it, even in the default mode.

The3DE, 2020-07-06:
The problem is that it sometimes probes too low, with SW mode off, it is usually low by about 1.3mm, and with SW mode on, it is usually by around .07mm. This means that while SW mode helps, it doesn’t fix the issue.

GMagician, 2020-07-06:
ok, thanks, there should no difference between SW or not SW mode seen on Marlin side when interrupts are on. It may be different when they are off (when non sw mode may miss some trigger) but now where does come's from your "0.07mm" too low? do you home and check nozzle real pos on the same point?
Sorry again but I need to correctly understand

[43 more replies]

closed 2020-07-21 by The3DE